OpenTabletDriver/OpenTabletDriver #577: Jitters between input and a certain area, unusable

OffxcialBubba, 2020-12-21:
## Description

When I first boot, the driver works fine. I only encounter problems when I start the driver, then it jitters randomly. It's centered around a weird area in the top left of my screen, unrelated to where my tablet area is.

https://user-images.githubusercontent.com/26052216/102790368-53bb0e80-4373-11eb-8a58-581e314bd5ff.mp4

The video shows osu! but it's persistent throughout the system.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Ubuntu 20.04 |
| Software Version | v0.4.2 |
| Tablet           | Wacom CTL-470 |

LavaDesu, 2020-12-21:
See the [Linux FAQ](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#i-have-a-ghost-cursor-constantly-alternating-cursor-position)

OffxcialBubba, 2020-12-21:
> See the [Linux FAQ](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#i-have-a-ghost-cursor-constantly-alternating-cursor-position)

Tried that, doesn't solve the issue

OffxcialBubba, 2020-12-21:
^ misclick

X9VoiD, 2020-12-21:
Blacklisting `wacom` module will only take effect after a reboot. You may do `sudo rmmod wacom` to force unload it without reboot.

InfinityGhost, 2020-12-21:
> When I first boot, the driver works fine.
This is the other kernel module being active. Closing because there's a known fix.

closed 2020-12-21 by InfinityGhost