OpenTabletDriver/OpenTabletDriver #583: Support Request: tablet not found (ArgumentOutOfRangeException)

Venji10, 2020-12-24:
# Support Request

I am trying to get my Gaomon S620 to work with OpenTabletDriver on Arch Linux. When running "otd", it searches for tablets, but throws an ArgumentOutOfRangeException for most tablets, including the S620 (Log below).

## What I've already tried

I tried fixing udev rules according to https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception .

## Diagnostic Information

error log: https://bpa.st/EE4Q

Help would be appreciated!

X9VoiD, 2020-12-24:
https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#argumentoutofrangeexception-value-0-15

Venji10, 2020-12-25:
Thank you, I overlooked that one, that fixed my issue.

closed 2020-12-25 by Venji10