OpenTabletDriver/OpenTabletDriver #586: Add support for Parblo a640

Sakamaio, 2020-12-26:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Parblo
| Product Name    | Parblo a640
| Width           |
| Height          | 609.6mm
| Pressure Levels | 8192
| LPI | 5080

These were hard to find so I can’t confirm if these are right.

## Diagnostic Information
 https://github.com/DIGImend/digimend-kernel-drivers/files/4555222/UC-Logic-ParbloA640-5543_0061.tar.gz
 

These were hard to find so I can’t confirm if these are right.

X9VoiD, 2020-12-26:
You're supposed to upload a diagnostics file that came from OTD itself, not from somewhere else.

OTD -> Help -> Export Diagnostics

InfinityGhost, 2020-12-26:
Reopening because it still isn't added to OpenTabletDriver

X9VoiD, 2021-01-03:
@Sakamaio just to clarify, download [OpenTabletDriver](https://github.com/InfinityGhost/OpenTabletDriver/releases/download/v0.4.2/OpenTabletDriver.win-x64.zip), extract it to any folder of your choice, open `OpenTabletDriver.UX.Wpf` and then click Help -> Export Diagnostics.

That should prompt you to name and save a file somewhere, pick any filename and location. Upload that file here.

X9VoiD, 2021-01-22:
@Sakamaio

Diffie-Hellman-str, 2021-01-28:
VendorID: 21827
[ParbloA640.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5888318/ParbloA640.txt)

X9VoiD, 2021-01-28:
@Diffie-Hellman-str please use the latest [OTD release](https://github.com/InfinityGhost/OpenTabletDriver/releases/tag/v0.5.0).

Diffie-Hellman-str, 2021-01-28:
Diagnostics file exported with the latest version
[ParbloA640V2.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5888593/ParbloA640V2.txt)

X9VoiD, 2021-01-28:
Extract [A640.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5888698/A640.zip) to `Configurations\Parblo\A640.json`, then restart OTD or press "Detect Tablet".

This configuration is expected to not work for now, but OTD should atleast detect it. Report back if it works/detected.

Diffie-Hellman-str, 2021-01-28:
@X9VoiD
Working:
 - The tablet detected
 - The cursor moves correctly
Does not work:
 - Changing the tablet area
 - All hotkeys, including those on the pen itself

X9VoiD, 2021-01-28:
Does your cursor look like a diamond/dot while hovering the pen?

Diffie-Hellman-str, 2021-01-28:
Like a diamond (standard cursor)

X9VoiD, 2021-01-28:
We will need to figure out the tablet's initialization procedure.

Download and install that tablet's OEM driver, also install [USBPcap](https://desowin.org/usbpcap/) then restart your computer.

Once booted again, make sure that your tablet is configurable through the OEM driver (stuff like changing tablet area). Go to `C:\Program Files\USBPcap` and run `USBPcapCMD.exe`

It'll prompt you which filter number to use for capturing USB packets. Find the filter number that corresponds to where your tablet is. Your tablet is most likely named `HID-compliant pen`.

Enter the correct filter number, then type any filename you want.

Once you enter a filename the capture will start, do the following

- Move pen for 5 secs
- Replug tablet
- Move pen for 5 secs

Close the window then upload the generated file here.

Diffie-Hellman-str, 2021-01-28:
I made 2 attempts just in case
[parbloA640.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5889454/parbloA640.zip)

X9VoiD, 2021-01-28:
[A640.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5889553/A640.zip)

Try this config

edichau, 2021-10-03:
i tried using this config, i cant seem to use it, i tried flipping orientations and no settings worked

jamesbt365, 2023-04-07:
Could somebody provide an up to date diagnostic for this tablet, I'm not sure if this fits under the v2 variant we have for this.

jamesbt365, 2023-08-23:
Closing in favour of the more up to date request #2829

closed 2023-08-23 by jamesbt365