OpenTabletDriver/OpenTabletDriver #589: Feature Request: Reload original drivers on close

emiliskiskis, 2020-12-26:
## Description

It would be very convenient if the original drivers were reloaded after closing OpenTabletDriver. I'm using a Huion HS64, which has native Windows Ink support and smooths input, it's more convenient for writing/drawing, though for osu! I want to use a smaller tablet area and no smoothing. When I close the program though, the original drivers aren't reloaded and I have to unplug the tablet and plug it back in for it to work again.

Mrcubix, 2020-12-26:
a plugin called windows ink is available [here ](https://github.com/X9VoiD/VoiDPlugins/releases/download/v0.2.1/WindowsInk.zip) (for 0.4.2, if you need it for a more recent version, then check [the action tab on that same repo](https://github.com/X9VoiD/VoiDPlugins/actions))
also use [this guide](https://github.com/X9VoiD/VoiDPlugins/wiki/WindowsInk) 
i would like to add that OTD's vmulti will only be used in artist mode (refer to the guide)

X9VoiD, 2020-12-30:
The correct term here is `PnP Windows Ink` mode, and the `Vendor Interface` mode.

All OTD does is send commands to the tablet to switch its mode to vendor interface (the same interface that non-default OEM drivers use) to be able to read input from the tablet.

There was no such thing as "OTD unloaded the original drivers". `PnP Windows Ink` *cannot* be unloaded, and tablets *cannot* be switched back to `PnP Windows Ink` mode since that functionality was never included in any tablet firmwares. From the OEMs POV there is no reason to revert back to the original mode if they have already switched the tablet's mode through their drivers, which OTD also does.

This is a `won't-fix` issue.

InfinityGhost, 2020-12-30:
For now, marking this wontfix unless we find a way to perform this elegantly.

closed 2020-12-30 by InfinityGhost