OpenTabletDriver/OpenTabletDriver #590: Support Request: Gaomon M10K Pro

zakaria-chahboun, 2020-12-26:
#### Hi!
First of all i want to say thanks to this project because it's really good
I hope you support **Gaomon M10K Pro**!

- lsusb: `256c:006d` (like Huion HS610)
- Vendor: `Gaomon`
- Product: `M10K Pro`
- Specs: `10″×6.25″, 5080 LPI, 266 RPS , 8192 PL, tilt-sensitive, Frame controls 10 buttons, touch ring with a center button`
- Os: Linux (pop!_Os)

Gaomon M10K Pro - https://www.gaomon.net/Pen_Tablet/M10KPRO.html

#### Image
[image]

#### Resources
[descriptors.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743612/descriptors.txt)
[frame_buttons.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743613/frame_buttons.txt)
[pen_buttons.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743614/pen_buttons.txt)
[pen_coords.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743615/pen_coords.txt)
[pen_pressure.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743616/pen_pressure.txt)
[pen_tilt.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743618/pen_tilt.txt)
[probe.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743619/probe.txt)
[touch_ring.anti-clockwise.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743620/touch_ring.anti-clockwise.txt)
[touch_ring.center_key.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743621/touch_ring.center_key.txt)
[touch_ring.clockwise.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743622/touch_ring.clockwise.txt)
[touch_ring.directions.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5743623/touch_ring.directions.txt)

Mrcubix, 2020-12-26:
an actual diagnostic from Inside OTD itself is required to make any config, in OpenTbaletDriver.UX.Wpf.exe, go to help > export diagnostic, and upload it here

zakaria-chahboun, 2020-12-28:
@Mrcubix 
This is my diagnostic file:

[diagnostics.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5748848/diagnostics.txt)

Mrcubix, 2020-12-28:
alright, i'll be working on it, coming back in 10 min

Mrcubix, 2020-12-28:
[M10K Pro.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5748931/M10K.Pro.zip)

zakaria-chahboun, 2020-12-30:
@Mrcubix This is the string `OEM02_T19n_200311` i will copy it in discrod #configs

gonX, 2021-03-18:
Select Artist Mode from the the drop down in the bottom left of the Output tab. Pressure will work with that enabled.

zakaria-chahboun, 2021-03-18:
@gonX That's true! Thanks!

InfinityGhost, 2021-03-20:
I remember merging a PR for this tablet, isn't this implemented now?

X9VoiD, 2021-03-20:
There's the M106K Pro, and then there's the M10K, without the Pro. This one is M10K Pro.

zakaria-chahboun, 2021-03-21:
This is the last `M10K Pro` configuration file from @X9VoiD  
[M10K_Pro.json.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6176995/M10K_Pro.json.zip)

zakaria-chahboun, 2021-03-25:
@InfinityGhost Why you don't add this tablet to project section https://github.com/OpenTabletDriver/OpenTabletDriver/projects/4

InfinityGhost, 2021-08-05:
This was resolved a while back, not sure why its still open.

closed 2021-08-05 by InfinityGhost