OpenTabletDriver/OpenTabletDriver #591: Add support for Acepen AP 1060

Custard7A, 2020-12-28:
# Tablet Information
| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Acepen |
| Product Name    | AP1060 |
| Width           | 10 Inches |
| Height          | 6 Inches |
| Pressure Levels | 8192 Levels |
| LPI             | 5080 LPI |

## Diagnostic Information
[Diagnostics-AP1060.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5746427/Diagnostics-AP1060.zip)

X9VoiD, 2020-12-28:
Make sure you are logged in to Github and then download latest [OTD dev build](https://github.com/InfinityGhost/OpenTabletDriver/actions?query=event%3Apush).

Extract the zip to any folder of your choice. Uninstall your previous OTD and delete `%localappdata%/OpenTabletDriver/settings.json` if it exists.

Redo the diagnostics using the OTD dev build so we can ease the process. The latest dev builds tells us which devices we are allowed to access which were not available on 0.4.2 diagnostics.

Custard7A, 2020-12-29:
Ok, the app reporting as v0.5.0 now.

[Diagnostics-AP1060-2.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5749603/Diagnostics-AP1060-2.zip)

X9VoiD, 2020-12-29:
Download and extract [AP 1060.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5749626/AP.1060.zip) as `Configurations\Acepen\AP 1060.json`

Restart OTD and see if it detects properly. Most of the contents of the config are dummy values for now and is made just for the sake of detection. Those values will be updated later once we got it detected in OTD.

Custard7A, 2020-12-29:
The "No tablet detected" message is gone and replaced by a representation of the tablet's area.

Under the Console tab in the log it includes:
[Detect:Info]	Searching for tablet 'Acepen AP 1060'
[Detect:Info]	Found tablet 'Acepen AP 1060'

When using the tablet to move the cursor around the cursor is the diamond dot, as it was before.

The Tablet Debugger window offers no data so far, when manipulating the tablet or otherwise.

X9VoiD, 2020-12-29:
I see, I'd recommend you join the [discord server](https://discord.gg/P3GYdUS). We will continue there, as the next steps require a lot of interaction. Or you may opt to not to and continue here instead.

Custard7A, 2020-12-30:
Following the progress made on Discord the other day, I've gone as far as I know how with attempting to verify the configuration .json you gave me for this tablet.

I have a fresh Windows PC that has been assembled newly this very month, and has not yet seen any real service. Theoretically a perfectly clean test-bed, but I began by attempting to verify that.

Checking 'Device Manager -> View -> Devices by connection' revealed nothing of any reference to pen tablets, or HIDs at all. This list is light on peripherals and I recognize all things on it, as expected of a fresh system.
Viewing by container showed my tablet as it was on the other PC, "TABLET 1060N". No sight of an "Uninstall" option for it.

I ran `pnputil -e` in Command Promp, went through the list looking for any mention of a Pen Tablet, finding none. Neither did I see one numbered "oem52.inf", or one I couldn't particularly account for.

So then running OTD; In short, it acted the same as left off on the other system. The interface responds to the tablet being there as if it is detected, and the cursor also changes from the diamond dot to the typical arrow pointer when using the tablet. However, the Tablet Debugger remains blank in all fields at all times, changing the size of the tablet area then pressing Apply had no effect, neither did attempts to change to Relative Mode, and pressure was working in both Photoshop and Krita without any plugin added to OTD.

Custard7A, 2021-01-30:
@X9VoiD Hello, I realize you probably didn't even notice I replied again here, since I didn't have the good thought to @ you back then. Sorry. I wanted to add that I haven't forgot about this, though a slow pace suite me, I'm happy to receive a next step in this place.

FlyMyPG, 2021-07-30:
What is the status of this driver?  I just got an Acepen AP960, and really want to stop using their Windows software.  I'm hoping the drivers are identical.

The info I've collected so far:

Brand: Acepen
Model: AP906
Dimensions: 263 x 197 x 9mm; 379 Grams
Reading Height: 10mm
Reading Speed: 266 P/S
Read Resolution: 5080 LPI
Precision: +/- 0.01mm
Drawing Area: 9 x 5 inches, 50809 x 25400 Points
Pressure: 8192 Levels
Tilt Range: +/- 60 Degrees

Linux work: https://githubmemory.com/repo/DIGImend/digimend-kernel-drivers/issues/540

Brand URL: https://www.ace-pen.com
Model URL: https://www.ace-pen.com/graphic-tablet/animation-digital-drawing-tablet/battery-free-animation-design-digital-drawing.html
Driver URL: https://www.ace-pen.com/support
Amazon URL: https://www.amazon.com/gp/product/B07VYNLMV8

Wireshark:

DEVICE DESCRIPTOR
    bLength: 18
    bDescriptorType: 0x01 (DEVICE)
    bcdUSB: 0x0110
    bDeviceClass: Device (0x00)
    bDeviceSubClass: 0
    bDeviceProtocol: 0 (Use class code info from Interface Descriptors)
    bMaxPacketSize0: 64
    idVendor: UC-Logic Technology Corp. (0x5543)
    idProduct: Unknown (0x0091)
    bcdDevice: 0x0000
    iManufacturer: 1
    iProduct: 2
    iSerialNumber: 3
    bNumConfigurations: 1

CONFIGURATION DESCRIPTOR
    bLength: 9
    bDescriptorType: 0x02 (CONFIGURATION)
    wTotalLength: 91
    bNumInterfaces: 3
    bConfigurationValue: 1
    iConfiguration: 0
    Configuration bmAttributes: 0x80  NOT SELF-POWERED  NO REMOTE-WAKEUP
    bMaxPower: 50  (100mA)
INTERFACE DESCRIPTOR (0.0): class HID
    bLength: 9
    bDescriptorType: 0x04 (INTERFACE)
    bInterfaceNumber: 0
    bAlternateSetting: 0
    bNumEndpoints: 1
    bInterfaceClass: HID (0x03)
    bInterfaceSubClass: Boot Interface (0x01)
    bInterfaceProtocol: Mouse (0x02)
    iInterface: 0
HID DESCRIPTOR
ENDPOINT DESCRIPTOR
    bLength: 7
    bDescriptorType: 0x05 (ENDPOINT)
    bEndpointAddress: 0x81  IN  Endpoint:1
    bmAttributes: 0x03
    wMaxPacketSize: 10
    bInterval: 3
INTERFACE DESCRIPTOR (1.0): class HID
    bLength: 9
    bDescriptorType: 0x04 (INTERFACE)
    bInterfaceNumber: 1
    bAlternateSetting: 0
    bNumEndpoints: 1
    bInterfaceClass: HID (0x03)
    bInterfaceSubClass: Boot Interface (0x01)
    bInterfaceProtocol: Mouse (0x02)
    iInterface: 0
HID DESCRIPTOR
ENDPOINT DESCRIPTOR
    bLength: 7
    bDescriptorType: 0x05 (ENDPOINT)
    bEndpointAddress: 0x82  IN  Endpoint:2
    bmAttributes: 0x03
    wMaxPacketSize: 10
    bInterval: 3
INTERFACE DESCRIPTOR (2.0): class HID
    bLength: 9
    bDescriptorType: 0x04 (INTERFACE)
    bInterfaceNumber: 2
    bAlternateSetting: 0
    bNumEndpoints: 2
    bInterfaceClass: HID (0x03)
    bInterfaceSubClass: No Subclass (0x00)
    bInterfaceProtocol: 0x00
    iInterface: 0
HID DESCRIPTOR
ENDPOINT DESCRIPTOR
    bLength: 7
    bDescriptorType: 0x05 (ENDPOINT)
    bEndpointAddress: 0x83  IN  Endpoint:3
    bmAttributes: 0x03
    wMaxPacketSize: 16
    bInterval: 3
ENDPOINT DESCRIPTOR
    bLength: 7
    bDescriptorType: 0x05 (ENDPOINT)
    bEndpointAddress: 0x03  OUT  Endpoint:3
    bmAttributes: 0x03
    wMaxPacketSize: 16
    bInterval: 10

Mrcubix, 2021-07-30:
Open your own issue, this one issue has nothing to do with your tablet

FlyMyPG, 2021-07-30:
No, you are wrong.  I'd have to uninstall the OEM driver to avoid a conflict with OTD.  It took me a while to get the OEM driver working well enough to use.  I'm not letting OTD anywhere near my system until my current work is done.  I can't afford the risk of things going wrong.  Please be patient.

jamesbt365, 2021-11-04:
Resolved in #1495

closed 2021-11-04 by jamesbt365