OpenTabletDriver/OpenTabletDriver #594: Support Request: can't detect gaomon s620

zienard, 2020-12-29:
# Support Request

just got this tablet and the driver can't detect it at all
## What I've already tried

I tried making a config based on the HIDs in my computer 
## Diagnostic Information

[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Huion H430P'
[Detect:Info]	Searching for tablet 'Huion H610 Pro V2'
[Detect:Info]	Searching for tablet 'Huion H610 Pro'
[Detect:Info]	Searching for tablet 'Huion H640P'
[Detect:Info]	Searching for tablet 'Huion HS610'
[Detect:Info]	Searching for tablet 'Huion HS611'
[Detect:Info]	Searching for tablet 'Huion HS64'
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[Detect:Info]	Searching for tablet 'Huion Q620M'
[Detect:Info]	Searching for tablet 'New Tablet'
[Detect:Info]	Searching for tablet 'SONiX USB DEVICE'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom CTH-470'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[Detect:Info]	Searching for tablet 'Wacom CTL-6100'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Wacom CTL-672'
[Detect:Info]	Searching for tablet 'Wacom CTL-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-690'
[Detect:Info]	Searching for tablet 'Wacom FT-0405-U'
[Detect:Info]	Searching for tablet 'Wacom PTH-451'
[Detect:Info]	Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom PTH-660'
[Detect:Info]	Searching for tablet 'Wacom PTH-850'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 03'
[Detect:Info]	Searching for tablet 'XP-Pen G430'
[Detect:Info]	Searching for tablet 'XP-Pen G430S'
[Detect:Info]	Searching for tablet 'XP-Pen G430S_B'
[Detect:Info]	Searching for tablet 'XP-Pen G540 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen G640 v2'
[Detect:Info]	Searching for tablet 'XP-Pen G640'
[Detect:Info]	Searching for tablet 'XP-Pen G640s'
[Detect:Info]	Searching for tablet 'XP-Pen G960 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Star 03v2'
[... 96 lines cut ...]
[Detect:Info]	Found tablet 'Logitech USB Receiver'
[NullReferenceException:Error]	Object reference not set to an instance of an object.
[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Huion H430P'
[Detect:Info]	Searching for tablet 'Huion H610 Pro V2'
[Detect:Info]	Searching for tablet 'Huion H610 Pro'
[Detect:Info]	Searching for tablet 'Huion H640P'
[Detect:Info]	Searching for tablet 'Huion HS610'
[Detect:Info]	Searching for tablet 'Huion HS611'
[Detect:Info]	Searching for tablet 'Huion HS64'
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[Detect:Info]	Searching for tablet 'Huion Q620M'
[Detect:Info]	Searching for tablet 'Logitech USB Receiver'
[Detect:Debug]	Using device 'USB Receiver'.
[Detect:Debug]	Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]	Device path: \\?\hid#vid_046d&pid_c539&mi_00#9&1b685b92&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
[Device:Error]	Failed to open stream.
[DeviceIOException:Error]	Unable to open HID class device (\\?\hid#vid_046d&pid_c539&mi_00#9&1b685b92&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd).
[Detect:Info]	Found tablet 'Logitech USB Receiver'
[NullReferenceException:Error]	Object reference not set to an instance of an object.

zienard, 2020-12-29:
my bad wrong label

X9VoiD, 2020-12-29:
I'm assuming you're using Windows. Make sure you've uninstalled every other tablet driver that you have. And also, check [wiki](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Windows-FAQ#my-gaomonhuion-tablet-isnt-detected-but-is-listed-as-supported).

zienard, 2020-12-29:
I've uninstalled every tablet driver and now it works thanks for the advice

closed 2020-12-29 by zienard