OpenTabletDriver/OpenTabletDriver #617: problem with filters

reodeza, 2021-01-02:
## Description

Hawku and Devocub filters doesnt showing in filters menu after installation them and restarting driver
[image]

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Win 10 1909
| Software Version | v0.4.2
| Tablet           | CTH-680

X9VoiD, 2021-01-02:
Make sure you're using the correct filter version, This [TabletDriverFilters](https://github.com/OpenTabletDriver/TabletDriverFilters/releases/tag/v0.1.0) release is compatible only with OTD 0.5.x. Check out #faq channel of Discord server for the OTD 0.4.2 compatible version.

Do note tho that the 0.4.2 compatible version have limitations, such as inability to fake report rate (so timer interval have to be set to tablet report rate), and slightly incorrect devocub implementation.

Use the 0.5.0 prerelease instead if you need any of the features mentioned above.

reodeza, 2021-01-02:
oh, ok thx, sorry for the opened issue

X9VoiD, 2021-01-03:
You may close this issue if you don't have any problems anymore.

InfinityGhost, 2021-01-06:
Issue appears resolved, closing.

closed 2021-01-06 by InfinityGhost