OpenTabletDriver/OpenTabletDriver #619: Support Request:

heagan01, 2021-01-02:
# Support Request
My opentabletdriver gui window in linux just keep crashing

https://user-images.githubusercontent.com/54394590/103461792-93123380-4d53-11eb-954e-23e9f6f89834.mp4

## What I've already tried
I've already tried restarting

## Diagnostic Information
This is the diagnostic when I ran it in the terminal
`
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
StreamJsonRpc.ConnectionLostException: The JSON-RPC connection with the remote party was lost before the request could complete.
   at StreamJsonRpc.JsonRpc.InvokeCoreAsync(JsonRpcRequest request, Type expectedResultType, CancellationToken cancellationToken)
   at StreamJsonRpc.JsonRpc.InvokeCoreAsync[TResult](RequestId id, String targetName, IReadOnlyList`1 arguments, IReadOnlyList`1 positionalArgumentDeclaredTypes, IReadOnlyDictionary`2 namedArgumentDeclaredTypes, CancellationToken cancellationToken, Boolean isParameterObject)
   at OpenTabletDriver.UX.MainForm.ResetSettings(Boolean force)
   at OpenTabletDriver.UX.MainForm.InitializeAsync()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.Gtk.MainClass.Main(String[] args)
`

My Gui crashed, so I can't export my diagnostics

X9VoiD, 2021-01-02:
Make sure that the Daemon is running. If it is, delete `~./config/OpenTabletDriver`, then restart Daemon and UX.

Also, read [FAQ](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ).

heagan01, 2021-01-03:
still doesn't work, i've tried removing the config folder, and restarting the daemon, and I made sure to start the Daemon, about the FAQ I think i've tried the "Neither Wayland nor X11 were detected" commands, before I've tried that, the gui works just fine, after I tried that, the gui crashed everytime I launched it

X9VoiD, 2021-01-03:
Don't run Daemon as sudo, same with UX. The [FAQ entry](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#daemon-crashes-with-neither-wayland-nor-x11-were-detected) hasn't been updated yet with new information, and you'll be affected if you haven't updated yet from the AUR or is using 0.4.2 deb package. The updated otd service's content is this:

```
[Unit]
Description=OpenTabletDriver Daemon
After=graphical.target
PartOf=graphical.target
StartLimitIntervalSec=60
StartLimitBurst=2

[Service]
ExecStart=/usr/bin/otd
Restart=always
RestartSec=10s

[Install]
WantedBy=graphical.target
```
If you installed OpenTabletDriver from a package like `.deb` or `AUR` then `systemctl --user restart opentabletdriver`

heagan01, 2021-01-04:
Thanks, the gui finally worked, but I have a separate question, how do I launch the daemon? When the system just launched?

X9VoiD, 2021-01-04:
It should already launch on boot if you enable the service.

`systemctl --user enable opentabletdriver`

closed 2021-01-04 by heagan01