OpenTabletDriver/OpenTabletDriver #621: Support Request: Cant start the tablet driver

rumeie, 2021-01-03:
# Support Request

[image]
after I click Start it show this 
[image]

and return back to 
[image]

https://user-images.githubusercontent.com/18748818/103478838-5f431680-4e04-11eb-8279-5e70c3881cac.mp4

## What I've already tried

Uninstall and reinstall the driver
## Diagnostic Information

after clicking Export Diagnostics generate this file

[daemon.log](https://github.com/InfinityGhost/OpenTabletDriver/files/5761604/daemon.log)

https://user-images.githubusercontent.com/18748818/103479002-599a0080-4e05-11eb-9d7c-e3a54d739767.mp4

X9VoiD, 2021-01-03:
Delete `%localappdata%/OpenTabletDriver/settings.json`

rumeie, 2021-01-03:
ok thx resolved

closed 2021-01-03 by rumeie