OpenTabletDriver/OpenTabletDriver #626: App crash on save

Azn9, 2021-01-05:
Every time I click the "save" button, the application crashes :(

**System** : Linux lite (Ubuntu 20.04)
**Tablet** : Gaomon S620
**Application version** : v0.4.2
**Error log** : 
```
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.Plugin.Log.Exception(Exception ex)
   at OpenTabletDriver.UX.MainForm.ApplySettings()
   at OpenTabletDriver.UX.MainForm.SaveSettings(Settings settings)
   at OpenTabletDriver.UX.MainForm.<ConstructMainControls>b__2_4(Object s, EventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.Gtk.MainClass.Main(String[] args)
```

InfinityGhost, 2021-01-06:
This should already be resolved in development builds, try this version: https://github.com/InfinityGhost/OpenTabletDriver/suites/1771455488/artifacts/33813797

If you need this as a deb package, let me know so I can package it and send across.

X9VoiD, 2021-01-22:
Obsolete issue

InfinityGhost, 2021-01-22:
> Obsolete issue

Closing, because it has since been resolved.

closed 2021-01-22 by InfinityGhost