OpenTabletDriver/OpenTabletDriver #629: OpenTabletDriver Two Monitors

Liedaazn, 2021-01-07:
I recently installed a new monitor and every time I use my tablet, the cursor will start teleporting between the spot my cursors at and a different place. It'll keep doing this unless I have hover just in the tablet range.

| Windows 10 |
| Tablet XP-PEN G430S_ |

X9VoiD, 2021-01-07:
Make sure you've uninstalled all other tablet drivers before using OTD.

Liedaazn, 2021-01-07:
> Make sure you've uninstalled all other tablet drivers before using OTD.

I have uninstalled all other tablet drivers, it worked previously before installing my second monitor.

X9VoiD, 2021-01-07:
G430/S config also got updated so download [.NET 5](https://dotnet.microsoft.com/download/dotnet/thank-you/runtime-5.0.1-windows-x64-installer) and try [OTD v0.5.0 pre-release](https://discord.com/channels/615607687467761684/615939905566015493/794446989739294780).

closed 2021-01-07 by Liedaazn