OpenTabletDriver/OpenTabletDriver #642: Support Request: Cursor only moves horizontally with G430S_B on Ubuntu

inc, 2021-01-13:
# Support Request

The tablet is detected and in the Tablet Debugger the horizontal position, buttons and pressure all work.

The vertical position is always stuck at either 0 or 32767.

"Position[5836,32767]"

The cursor only moves horizontally on the top or bottom of the screen.

Have tried the G430S_B on Windows with the official drivers and everything works.

Any idea why it's stuck?

Thanks.

[diagconfig.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5807833/diagconfig.zip)

InfinityGhost, 2021-01-20:
This is a really odd situation, really no clue why it has 32767 as a value at all, thats the upper limit for position on any digitizer.

inc, 2021-01-20:
A quick update. If I try using the tablet after some long period of time (the next day) everything works perfectly. If I unplug it and plug it back in, the issue returns for at least several hours. I also see the same top/bottom issue with the XP-PEN Linux Beta driver so it's probably a problem with Ubuntu or my hardware and not a problem specific to OpenTabletDriver.

InfinityGhost, 2021-01-20:
That does seem to be the case, seems like the firmware of your tablet might have a bug that causes that. I'm gonna close this since it appears to be caused by something other than OpenTabletDriver.

closed 2021-01-20 by InfinityGhost

zer0hi, 2021-04-24:
I had simillar problem in Debian, Ubuntu and now in Manjaro. In three different computers. Cursor moves only horizontally on a top on the screen. Any drivers to this tablet has this problem. But once in Debian with official drivers this tablet started to work on a two weeks, that's strange