OpenTabletDriver/OpenTabletDriver #646: Support Request: Where can I find v0.5 ?

UsaidMuhammad, 2021-01-13:
# Support Request
Need to download v0.5

## What I've already tried
Checked both OTD released and its installer releases no 0.5 tags or release to be found
[image]

[image]

gonX, 2021-01-13:
Under [Actions](https://github.com/InfinityGhost/OpenTabletDriver/actions). 0.5.0 hasn't been finalized yet.

UsaidMuhammad, 2021-01-13:
Apologies for my incompetence, but may I request a few more hand-holding as I am unable to find a place in action to get the v0.5 build.

Do I have to clone master or action and build it myself?

X9VoiD, 2021-01-13:
Prebuilts for 0.5.0 prerelease can be found on the official [OTD discord server](https://discord.com/channels/615607687467761684/615939905566015493/794446989739294780).

gonX, 2021-01-13:
> I am unable to find a place in action to get the v0.5 build
From that site you have to click a commit for which artifacts have been generated. Once you're on a commit page, there should be artifacts listed at the bottom with links to the precompiled builds.

I suggest using the pre-release files found on the OTD discord server as suggested by X9VoiD above, as they have the added bonus of being more widely used, so bugs are more predictable.

UsaidMuhammad, 2021-01-14:
Thanks for the support!

closed 2021-01-14 by UsaidMuhammad