OpenTabletDriver/OpenTabletDriver #647: Huion NEW1060 Plus 8192 pressure model does not detect

CyraMayCry, 2021-01-14:
## Description
Huion NEW1060 Plus 8192 pressure model is not being detected by the software

## System Information:
| Operating System | Windows 10 64 bit
| Software Version | OpenTabletDriver v0.4.1
| Tablet           | Huion NEW1060 Plus

CyraMayCry, 2021-01-14:
wait the software version is 0.4.2

Mrcubix, 2021-01-14:
a diagnostic of the tablet via OTD is required and asked for in the template, please provide that piece of information before we can proceed

CyraMayCry, 2021-01-14:
{
  "App Version": "OpenTabletDriver v0.4.2",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19041.0",
    "VersionString": "Microsoft Windows NT 10.0.19041.0"
  },
  "Environment Variables": {
    "USERDOMAIN": "DESKTOP-ES4H99K",
    "FPS_BROWSER_APP_PROFILE_STRING": "Internet Explorer",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "PROCESSOR_IDENTIFIER": "Intel64 Family 6 Model 158 Stepping 10, GenuineIntel",
    "Path": "C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Users\\User\\AppData\\Local\\Microsoft\\WindowsApps;",
    "SESSIONNAME": "Console",
    "GOOGLE_DEFAULT_CLIENT_ID": "no",
    "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-ES4H99K",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "ComSpec": "C:\\WINDOWS\\system32\\cmd.exe",
    "OS": "Windows_NT",
    "TMP": "C:\\Users\\User\\AppData\\Local\\Temp",
    "ProgramFiles": "C:\\Program Files",
    "OneDrive": "C:\\Users\\User\\OneDrive",
    "OneDriveConsumer": "C:\\Users\\User\\OneDrive",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "GOOGLE_DEFAULT_CLIENT_SECRET": "no",
    "PROCESSOR_REVISION": "9e0a",
    "ProgramW6432": "C:\\Program Files",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "USERNAME": "User",
    "PUBLIC": "C:\\Users\\Public",
    "HOMEPATH": "\\Users\\User",
    "COMPUTERNAME": "DESKTOP-ES4H99K",
    "TEMP": "C:\\Users\\User\\AppData\\Local\\Temp",
    "ProgramData": "C:\\ProgramData",
    "LOGONSERVER": "\\\\DESKTOP-ES4H99K",
    "USERPROFILE": "C:\\Users\\User",
    "FPS_BROWSER_USER_PROFILE_STRING": "Default",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "SystemDrive": "C:",
    "APPDATA": "C:\\Users\\User\\AppData\\Roaming",
    "PROCESSOR_LEVEL": "6",
    "HOMEDRIVE": "C:",
    "SystemRoot": "C:\\WINDOWS",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "windir": "C:\\WINDOWS",
    "NUMBER_OF_PROCESSORS": "6",
    "LOCALAPPDATA": "C:\\Users\\User\\AppData\\Local",
    "GOOGLE_API_KEY": "no",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "__COMPAT_LAYER": "DetectorsAppHealth",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#hid&col03#1&4784345&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 255,
      "ProductID": 47820,
      "ReleaseNumberBcd": 1,
      "ReleaseNumber": "0.1",
      "ProductVersion": 1,
      "Manufacturer": "DJP Inc.",
      "ProductName": "Virtual Multitouch Device",
      "SerialNumber": "123123123",
      "MaxInputReportLength": 5,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
    {
[... 760 lines cut ...]
      "Time": "2021-01-15T02:42:15.7309051+08:00",
      "Group": "Settings",
      "Message": "Tip Binding: 'Mouse Button Binding: Left'@1%",
      "Level": 1
    },
    {
      "Time": "2021-01-15T02:42:15.7309123+08:00",
      "Group": "Settings",
      "Message": "Pen Bindings: [0, ], [1, ]",
      "Level": 1
    },
    {
      "Time": "2021-01-15T02:42:15.7309186+08:00",
      "Group": "Settings",
      "Message": "Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]",
      "Level": 1
    },
    {
      "Time": "2021-01-15T02:42:15.7309249+08:00",
      "Group": "Settings",
      "Message": "Driver is auto-enabled.",
      "Level": 1
    }
  ]
}

CyraMayCry, 2021-01-14:
this one?

Mrcubix, 2021-01-14:
delete winUSB then huion drivers in that order. reinstall if winUSB is still present.

after that:

Device Manager -> View -> Devices by Container
Find for any of these devices:
  -  Gaomon HID/Tablet
  -  Huion HID/Tablet
  -  Graphics Tablet Device
  -  HID Tablet
Expand drop down then right click on matching device
Click Update -> "Browse my computer" -> "Let me pick" -> HID-compliant device or USB Input Device

install https://dotnet.microsoft.com/download/dotnet/thank-you/runtime-desktop-5.0.2-windows-x64-installer
then update to that build https://github.com/InfinityGhost/OpenTabletDriver/suites/1831266993/artifacts/35353847

CyraMayCry, 2021-01-15:
How do i find winUSB and the huion drivers location in the directory?

Mrcubix, 2021-01-15:
like any program, in "change or uninstall a program"

CyraMayCry, 2021-01-15:
same with huion drivers?

CyraMayCry, 2021-01-15:
nvm But theres no either of the devices as the one u listed above

Mrcubix, 2021-01-15:
just proceed with the next steps which is installing the right dotnet and the pre-release

install https://dotnet.microsoft.com/download/dotnet/thank-you/runtime-desktop-5.0.2-windows-x64-installer
then update to that build https://github.com/InfinityGhost/OpenTabletDriver/suites/1831266993/artifacts/35353847

CyraMayCry, 2021-01-15:
ok it works now thx

Mrcubix, 2021-01-15:
you may close this issue now

CyraMayCry, 2021-01-15:
can i still use this driver if huion driver is installed?

Mrcubix, 2021-01-15:
No

CyraMayCry, 2021-01-15:
aw thats bummer ok then

closed 2021-01-15 by CyraMayCry