OpenTabletDriver/OpenTabletDriver #653: Support Request: CTL-472 not detecting, previously did

mazziechai, 2021-01-16:
# Support Request
Hello! I use a CTL-472, and recently the tablet driver has stopped detecting the tablet despite it working perfectly fine before just two days ago.

## What I've already tried
Restarting computer, reinstalling OpenTabletDriver, uninstalling the built in Windows drivers for the tablet. Please keep in mind that this is a brand new tablet; I've only used it for about a few days until this happened, so I'd like to believe that it's not the tablet's fault.

## Diagnostic Information
https://paste.gg/p/anonymous/743a469ef7ba432798c11f1104628ebe

X9VoiD, 2021-01-16:
You most likely played around with tablet configuration since OTD searches for a non-existent tablet. Reinstall OTD.

X9VoiD, 2021-01-16:
Just to be entirely sure, open a cmd prompt and run the following command

```
taskkill /IM OpenTabletDriver.Daemon.exe /F >nul 2>nul
taskkill /IM OpenTabletDriver.UX.Wpf.exe /F >nul 2>nul
rmdir /S /Q "%localappdata%\OpenTabletDriver"
```
and then you may install OTD again

mazziechai, 2021-01-16:
> 
> 
> You most likely played around with tablet configuration since OTD searches for a non-existent tablet. Reinstall OTD.

This was the issue, thanks. I feel silly now.

closed 2021-01-16 by mazziechai