OpenTabletDriver/OpenTabletDriver #654: Add support for

skoaned, 2021-01-16:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer   | Wacom |
| Product Name    | Wacom CTL6100WL | 
| Width           | 216.0mm |
| Height          | 135.0 mm |
| Pressure Levels | 4096 |
| LPI             | 2540 |

## Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.4.2",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19041.0",
    "VersionString": "Microsoft Windows NT 10.0.19041.0"
  },
  "Environment Variables": {
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "LOCALAPPDATA": "C:\\Users\\admin\\AppData\\Local",
    "TMP": "C:\\Users\\admin\\AppData\\Local\\Temp",
    "PROCESSOR_LEVEL": "6",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "NUMBER_OF_PROCESSORS": "8",
    "PROCESSOR_IDENTIFIER": "Intel64 Family 6 Model 60 Stepping 3, GenuineIntel",
    "GOOGLE_DEFAULT_CLIENT_SECRET": "no",
    "Path": "C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Program Files\\Java\\jdk-12\\bin;C:\\Program Files\\Git\\cmd;C:\\Program Files\\dotnet\\;C:\\Users\\admin\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Program Files\\Java\\jdk-12\\bin;C:\\Users\\admin\\.dotnet\\tools",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "USERDOMAIN": "DESKTOP-KK6100N",
    "LOGONSERVER": "\\\\DESKTOP-KK6100N",
    "windir": "C:\\Windows",
    "SESSIONNAME": "Console",
    "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-KK6100N",
    "COMPUTERNAME": "DESKTOP-KK6100N",
    "HOMEDRIVE": "C:",
    "ProgramFiles": "C:\\Program Files",
    "OneDrive": "C:\\Users\\admin\\OneDrive",
    "ComSpec": "C:\\Windows\\system32\\cmd.exe",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\Windows\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "OS": "Windows_NT",
    "APPDATA": "C:\\Users\\admin\\AppData\\Roaming",
    "__COMPAT_LAYER": "DetectorsAppHealth",
    "SystemRoot": "C:\\Windows",
    "TEMP": "C:\\Users\\admin\\AppData\\Local\\Temp",
    "USERPROFILE": "C:\\Users\\admin",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "GOOGLE_API_KEY": "no",
    "USERNAME": "admin",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "SystemDrive": "C:",
    "ProgramW6432": "C:\\Program Files",
    "HOMEPATH": "\\Users\\admin",
    "PUBLIC": "C:\\Users\\Public",
    "ProgramData": "C:\\ProgramData",
    "PROCESSOR_REVISION": "3c03",
    "GOOGLE_DEFAULT_CLIENT_ID": "no"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c539&mi_02&col03#7&6ed728e&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 1133,
      "ProductID": 50489,
      "ReleaseNumberBcd": 14598,
[... 952 lines cut ...]
      "Time": "2021-01-16T11:23:35.2000024-05:00",
      "Group": "Detect",
      "Message": "Using auxiliary device 'Intuos BT M'.",
      "Level": 0
    },
    {
      "Time": "2021-01-16T11:23:35.2000081-05:00",
      "Group": "Device",
      "Message": "A device stream was disposed.",
      "Level": 0
    },
    {
      "Time": "2021-01-16T11:23:35.2000139-05:00",
      "Group": "Detect",
      "Message": "Using auxiliary report parser type 'AuxReportParser'.",
      "Level": 0
    },
    {
      "Time": "2021-01-16T11:23:35.2000197-05:00",
      "Group": "Detect",
      "Message": "Device path: \\\\?\\hid#vid_056a&pid_0378&col02#6&3addac5b&1&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Level": 0
    }
  ]
}

X9VoiD, 2021-01-22:
As shown in diagnostics your tablet was detected properly. CTL-6100WL is supported. What is your actual issue?

gonX, 2021-12-01:
Closing as stale issue given the tablet is supported. Please open a new support request issue if you have issues getting your tablet to work.

closed 2021-12-01 by gonX