OpenTabletDriver/OpenTabletDriver #658: Can't install the plugin.

qqLimbo, 2021-01-17:
## Description
[Plugin:Error]	Failed to get one or more types. The plugin 'DevocubFilters' is likely out of date.

## System Information:

| Name bug           | Value |
| ---------------- | ----- |
| Operating System  Windows 10 x64 20H2 |
| Software Version last version |
| Tablet    wacom-472       |

Mrcubix, 2021-01-17:
not latest version, download and install https://dotnet.microsoft.com/download/dotnet/thank-you/runtime-desktop-5.0.2-windows-x64-installer
https://cdn.discordapp.com/attachments/615939905566015493/794447041765834782/OpenTabletDriver.win-x64.zip
go to plugin > plugin manager and download compatible plugins there

qqLimbo, 2021-01-17:
I can't run your OTD cuz says app should not run with admin priveleges, but i don't run with adm priv. So i download version v0.4.2 from discord server and updated .NET. OpenTabletDriverPlugins.dll works... but still same problem [Plugin:Error] with DevocubFilters or Kuuube-s-CHATTER-EXTERMINATOR. I have midnight, so i go to sleep, ty for any help btw.

Mrcubix, 2021-01-17:
if it prompt to not run the app as admin then your system run apps as admin by default, re-enable UAC and if that's not possible cause you used a debloater, then reinstall windows and setup debloater to not remove UAC

InfinityGhost, 2021-01-20:
This issue belongs on the TabletDriverFilters repository.

closed 2021-01-20 by InfinityGhost