OpenTabletDriver/OpenTabletDriver #662: Unable to detect the tablet

cl20103B11, 2021-01-18:
## Description
Console just show this error
 `[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')`
[image]

and why the diagnostics is a blank file??

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Pop!_OS 20.10 |
| Software Version | v0.5.0 |
| Tablet           | XP-Pen G430S |

X9VoiD, 2021-01-18:
https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ

stllok, 2021-01-18:
humm....
```
echo "blacklist hid-uclogic" | sudo tee -a /etc/modprobe.d/blacklist.conf
sudo rmmod hid-uclogic
```

still have a same error after run that script

[image]

and seem it didn't detect G430S?

btw I can't run opentabletdriver on root or sudo
```
root@pop-os:~# opentabletdriver
Failed to connect to bus: No such file or directory
Failed to connect to bus: No such file or directory

(OpenTabletDriver.UX.Gtk:8170): libnotify-WARNING **: 21:50:54.354: Failed to connect to proxy
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.UX.MainForm.DetectAllTablets()
   at OpenTabletDriver.UX.MainForm.<ConstructMenu>b__4_7(Object sender, EventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.App.Run(String platform, String[] args)
   at OpenTabletDriver.UX.Gtk.Program.Main(String[] args)
```
```
stllok@pop-os:~$ sudo opentabletdriver
Failed to connect to bus: No such file or directory
Failed to connect to bus: No such file or directory

(OpenTabletDriver.UX.Gtk:8567): libnotify-WARNING **: 21:52:19.629: Failed to connect to proxy
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.UX.MainForm.DetectAllTablets()
   at OpenTabletDriver.UX.MainForm.<ConstructMenu>b__4_7(Object sender, EventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.App.Run(String platform, String[] args)
   at OpenTabletDriver.UX.Gtk.Program.Main(String[] args)
```

X9VoiD, 2021-01-18:
You're supposed to restart the daemon after running the script. And **do not** run OTD as root.

stllok, 2021-01-18:
restarted, and still got that error

X9VoiD, 2021-01-18:
Just the daemon lol. If you rebooted the whole system you'll have to unload the module again, or add it to blacklist so you won't have to do it every reboot. Read FAQ.

stllok, 2021-01-21:
following the second and fifth FAQ and install the xp-pen linux driver, now it can detected. But it detecd as G540 not G430S
[image]

Weird

X9VoiD, 2021-01-21:
@stllok Open OTD -> Tablet -> Device string reader

```String Index: 4```

Then hit `Send Request`, paste output here.

stllok, 2021-01-21:
V2.0 20171204

X9VoiD, 2021-01-21:
Try to get String Index from 0 to 10

stllok, 2021-01-21:
0: Љ
1: XP-PEN
2: G430S
3-10 : V2.0 20171204

X9VoiD, 2021-01-30:
Extract [XP-Pen.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5897371/XP-Pen.zip) to `Configurations/XP-Pen`, then press `Detect tablet` and see if that fixes the G430/S/G540 mix up.

stllok, 2021-02-01:
[image]

G430 (sorry for the late)

lululock71, 2021-02-05:
I just had a similar issue (Kamvas Pro 13 not detected) and running
`echo "blacklist hid-uclogic" | sudo tee -a /etc/modprobe.d/blacklist.conf
sudo rmmod hid-uclogic
`
solved my issue too. My tablet is recognized but does not work correctly (but that's for another issue haha).

stllok, 2021-02-07:
> I just had a similar issue (Kamvas Pro 13 not detected) and running
> `echo "blacklist hid-uclogic" | sudo tee -a /etc/modprobe.d/blacklist.conf sudo rmmod hid-uclogic `
> solved my issue too. My tablet is recognized but does not work correctly (but that's for another issue haha).
> 
> I have to run it at each reboot tho, I dunno why.

humm... I can detect the tablet now, but now the problem is it did not detect correctly lol

Mrcubix, 2021-02-07:
what do you mean by not detected correctly? also paste he new console logs here and a diagnostic file from help > export diag

InfinityGhost, 2021-02-18:
Issue appears stale, if there's no update in 7 days it'll be closed.

stllok, 2021-02-25:
I think it can close now since the initial problem is fixed, the only problem is #698 for now

closed 2021-02-25 by stllok