OpenTabletDriver/OpenTabletDriver #669: Add support for XP-Pen Artist Pro 12

Izerpizer, 2021-01-18:
# Tablet Information

| Name XP-Pen Artist Pro 12 | Value 249.99 USD |
| --------------- | ----- |
| Manufacturer  XP-Pen |
| Product Name  Artist Pro 12 |
| Width          13.83" |
| Height         8.87" |
| Pressure Levels 8192 |
| LPI            I'm not sure what this is |

## Diagnostic Information

I'm not sure what this is talking about

Mrcubix, 2021-01-18:
For diag, in OpenTabletDriver.UX go to help > export diagnostic

Izerpizer, 2021-01-18:
@Mrcubix Do you just want the device information for the tablet, or would you like me to post the full contents  of the file?

Izerpizer, 2021-01-18:
I'm going to guess its the full file

Mrcubix, 2021-01-18:
full file ofc

Izerpizer, 2021-01-19:
@Mrcubix it has now been added.

Izerpizer, 2021-01-19:
@Mrcubix out of curiosity, how long do you think it would take to add support for this XP-Pen tablet?

X9VoiD, 2021-01-19:
@Izerpizer use the latest release (0.5.0) for getting the diagnostics and uploading the diagnostics as a file is preferred than copy pasting its contents.

Izerpizer, 2021-01-19:
@X9VoiD do I have to manually update to 0.5? I did "sudo apt update" and the OpenTabletDriver Package was still listed as I think it was 0.4.2.

X9VoiD, 2021-01-19:
https://github.com/InfinityGhost/OpenTabletDriver/releases/tag/v0.5.0

Izerpizer, 2021-01-20:
@X9VoiD  so i installed OpenTabletDriver 0.5.0, and it instantly crashes when i try to open the gui

Izerpizer, 2021-01-20:
@X9VoiD @Mrcubix , i have updated the diagnostic information, and uploaded it in file format.

side note, out of curiosity, in the installation guide, it doesnt say how to add open tablet driver as a repository for my package manager. So if i want to update the driver i seemingly have to manually update it.

X9VoiD, 2021-01-22:
Extract [Artist Pro 12.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5855291/Artist.Pro.12.zip) into `Configurations\XP-Pen\Artist Pro 12.json` and see if it gets detected.

Izerpizer, 2021-01-23:
@X9VoiD Alright I added the configuration as you described and it did initially get detected; however, I rebooted, and all of a sudden it was not able to detect the tablet any more. The `Artist Pro 12.json` file is still in the configuration folder after reboot, and the console shows that OpenTabletDriver tried searching for it, but it wasn't able to detect it. It's strange, it detected it fine at first, and then the driver crashed, so I rebooted, and now it doesn't detect it at all even though I have changed nothing.

closed 2021-01-29 by InfinityGhost