OpenTabletDriver/OpenTabletDriver #674: Inverted aspect ratio, and locked area location

NotMizxy, 2021-01-19:
## Description
When I first installed (not updated) v0.5.0 and tried to set my area instead of having the aspect ratio what it should be it inverted and appeared to be at a 90degree rotation. After a couple restarts the issue no longer occurred, but a new issue came up that locked the where my area was to either the left edge or the right edge, I was unable to move it away from the edge.  Both issues went away after a couple restarts and a combination of toggling different settings on and off repeatedly with restarts/saves/applies in between. 

## System Information:
| Name             | Value |
| OpenTabletDriver | v0.5.0 |
| Windows 10 |
| v0.5.0 |
| Gaomon S620  |

X9VoiD, 2021-01-22:
Please provide a clear and defined steps to reproduce the error so we may debug and fix it.

InfinityGhost, 2021-02-04:
Retry this on v0.5.1, still have yet to see anyone else with this issue.

NotMizxy, 2021-02-05:
Alright, I will try to replicate it, I didn't do anything special during the first install so I'll just install normally.

InfinityGhost, 2021-02-18:
Issue appears to be stale, will close in 7 days if nothing comes of it.

closed 2021-05-24 by X9VoiD