OpenTabletDriver/OpenTabletDriver #681: OpenTabletDriver should not be run with administrator privileges

mazzu, 2021-01-20:
I'm on Windows and using the only user i have set up, which has administrator privileges. When opening the UI (not the daemon or the installer) I get this error "OpenTabletDriver should not be run with administrator privileges".

I looked in every .exe's file properties and "run as administrator" is unchecked under the "compatibility" tab.
This is happening since 0.5.0 update.

[image]

jjhhgg100123, 2021-01-20:
Send us a picture of the UAC Control Settings page, just for confirmation. It has a slider to adjust the setting.

KissMyUSSR, 2021-01-20:
Just got exactly the same problem.

mazzu, 2021-01-20:
I too have the option set to "never notify"
[image]

mazzu, 2021-01-20:
I too have the option set to "never notify". 

[image]

Changing to default fixed it, but anyway it doesn't prompt me anything so I don't get it. Also what if I want to leave it to "never notify"?

KissMyUSSR, 2021-01-20:
Yup, changing it to default fixed it but I want to keep it on never. So, could you please fix that. Very appreciated for your work.

AbstractQbit, 2021-01-20:
Having UAC disabled is essentially equivalent to running absolutely everything as root on UNIX-like systems. If the latter is frowned upon as being a security nightmare, why would that not be the case in Windows world?

Anyway, that error will probably get swapped to a warning in an upcoming hotfix, but you still shouldn't have UAC disabled regardless.

mazzu, 2021-01-20:
In fact I was just second-guessing my choice, I don't know enough of Windows about user permissions, also I thought that only executables which explicitly runs as admin were given special privileges, not everything. If that's the case maybe the best practice is using a normal user on windows like it's done on UNIX?

Anyway I digress, thank you for the awesome work!

AbstractQbit, 2021-01-20:
Back in Windows XP, if your account was an admin, everything you ran had all administrative permissions. That was a problem because people used those accounts for everyday stuff, when that was not what they were designed for. 

In Windows Vista Microsoft added UAC to tackle that giant security hole. What it does is it makes a second unpriviledged account for every administrative one behind the scenes, and runs programs with it. If a program depends on administrative access, then UAC prompts user whether that program is trusted enough to be ran as admin.

By disabling that feature everything starts working the way it used to in XP days, ie really insecurely.

And yes, you are absolutely correct, having a separate user account for everyday use and elevating to admin by typing a password in UAC is indeed the safest way to use your system.

InfinityGhost, 2021-01-20:
FWIW, we're gonna change this prompt just to warn users that there are potential issues with OTD by running without UAC or as an administrator.

closed 2021-01-21 by InfinityGhost

adxv, 2021-02-12:
Workaround: Create a bat file with the following command:
runas /trustlevel:0x20000 C:\Users\adams\Downloads\OpenTabletDriver.Installer.exe

PR #695 (2021-01-21): Warn users about running as admin instead of force closing
closes #681