OpenTabletDriver/OpenTabletDriver #682: I put the plugin into plugin folder doesnt appear error

schbs, 2021-01-20:
## Description

Basically when I put the plugin into plugin folder, but its still doesnt view in OpenTabletDriver App (sorry about my english)
## System Information:

| Name   Ger          | Value |
| ---------------- | ----- |
| Operating System | Windows 10
[image]
[image]

| Software Version | 0.4.1
| Tablet           | Intuos S Wacom

InfinityGhost, 2021-01-20:
The plugin installation method has changed, you would drag and drop the dll into the plugin manager window. However, this does bring to my attention a bug revolving around showing manually installed plugins.

InfinityGhost, 2021-01-20:
Going to open a separate issue for that bug.

closed 2021-01-20 by InfinityGhost