OpenTabletDriver/OpenTabletDriver #704: Huion NEW 1060PLUS (8192) not detected

HoutarouOreki, 2021-01-22:
## Description
Even though it's listed as supported, it's not being detected. I've tried uninstalling it in device manager, uninstalling the HUION software and restarting my PC.
It is the 8192 version, not 2048, so it's not the same as #350.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 Pro 1909
| Software Version | v0.5.0
| Tablet           | Huion NEW 1060PLUS (8192)

X9VoiD, 2021-01-22:
OTD -> Help -> Export Diagnostics. Upload the file generated here.

HoutarouOreki, 2021-01-22:
It started working when I turned the software on to generate the diagnostics, even though I didn't do anything else? Weird but I guess I'll just close this issue.

closed 2021-01-22 by HoutarouOreki