OpenTabletDriver/OpenTabletDriver #707: Cursor not moving in osu

Fxsch, 2021-01-22:
## Description

The cursor is moving perfectly fine on my desktop and in every other application except for osu! and osu! lazer. When I set the game to windowed, move my cursor outside the window and then move it back inside osu! with my tablet, it works aswell but it won't work in fullscreen or in borderless mode. It also works in other programs when they're in fullscreen, just not any of the two osu! versions.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Manjaro Gnome, Gnome version 3.38.3, newest Manjaro version
| Software Version | 0.5.0.r40.ge7bd0e5-1
| Tablet           | Veikk S640

X9VoiD, 2021-01-22:
OTD -> Help -> Export Diagnostics

Fxsch, 2021-01-22:
[Diagnostics.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5857291/Diagnostics.txt)

Fxsch, 2021-01-22:
After switching between fullscreen, borderless and windowed a few times it suddenly works as long as I don't go out of the window but as soon as I go out of osu and into osu again it stops working

Arthur095, 2021-01-25:
Try to not launch game in admin mode if you did.

ghdht, 2021-01-31:
Same issue on Lazer.
[Diagnostics.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5900505/Diagnostics.txt)
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Arch Gnome 3.38.3
| Software Version | v0.5.1-r0.gbccf6af
| Tablet           | CTH-480

Mrcubix, 2021-01-31:
don't run osu as admin, re-enable UAC to the 2nd top level, disable all option related to raw input ingame and you should be set

ghdht, 2021-02-02:
Only happens on wayland for me

InfinityGhost, 2021-02-03:
Odd that it only happens on wayland, since we're using libevdev. Maybe actually a lazer issue?

Fxsch, 2021-02-08:
In osu! lazer it is possible to move the cursor on the second screen and then back onto the main screen with the tablet to make it work but osu! stable can't be the "selected" application to make it work. If I switch the window to something else and still have osu! opened, I can also move the cursor in osu! with my tablet but as soon as I click on it and osu! becomes the selected window again, I can only move it with my mouse

X9VoiD, 2021-02-08:
I think this is an issue outside OTD's scope, as OTD uses `/dev/uinput` for reporting input and if OTD miscommunicates something, then it shouldn't be working for all linux users. The input reports are then to be consumed by `libinput` so there's nothing that OTD can do if something happens between `/dev/uinput` and `libinput`.

Fxsch, 2021-02-18:
I found a workaround: If osu! and OpenTabletDriver are both opened, press the Windows key and select OpenTabletDriver, osu! will then put itself in the foreground anyways and the cursor can be moved with the tablet

Fxsch, 2021-02-23:
After switching to Manjaro with KDE, which doesn't use Wayland, it works perfectly fine

InfinityGhost, 2021-02-23:
Really doesn't make sense why this only happens on Wayland, unless SDL2 is at fault here

VeryLongNicknameSuchWow, 2021-03-07:
Having the same issue on Gnome 3.38.3 with Wayland. Under X11 everything works fine.
Only lazer is affected, standard is fine on both.

VeryLongNicknameSuchWow, 2021-03-24:
This is now fixed with the OTD integrated into osu! Works fine on wayland.

osufan11, 2021-04-08:
> ## Description
>
> The cursor is moving perfectly fine on my desktop and in every other application except for osu! and osu! lazer. When I set the game to windowed, move my cursor outside the window [...]

Try turning off High Precision Mouse in osu settings.

osufan11, 2021-04-08:
> This is now fixed with the OTD integrated into osu! Works fine on wayland.

If it is fixed the author  or a member of the organization should close this.

closed 2021-04-08 by X9VoiD

GreenTeaSeb, 2022-04-01:
Having the same issues right now on arch wayland with wine osu. Lazer works.

jamesbt365, 2022-04-01:
may sure the raw input is off, if this is one specific application and not everything then this isn't an otd issue.

GreenTeaSeb, 2022-04-01:
yep, raw input is off. And you're right, it's not an OTD issue. The pen input works for clicking, but not more movement. While the default kernel drivers work. 
The moment I focus into the app with OTD it stops moving the pen, only registering the clicks.

Valoriks, 2022-05-05:
> Interestingly enough, only happens on absolute mode, artist mode works.

You're right, it works when I switch to artist mode! Thanks! I can finally play osu! on Wayland

jamesbt365, 2025-07-20:
Dismissed? this is not our issue, we cannot fix it or do anything about it. Just use artist mode, its now the default output mode and is better supported on wayland because most compositors can handle tablet input, but not so many have implemented support for absolute mice.