OpenTabletDriver/OpenTabletDriver #713: Wacom CTH-480 randomly stopped working

iceflarexd, 2021-01-24:
Making this for a friend of mine, as my tablet is working fine but I'm trying to troubleshoot his. He has a CTH-480 that was working perfectly with these drivers for a month or so. However, as of yesterday, the tablet driver no longer lets him move his cursor or anything. In the tablet debugger, he can see the tablet's input being detected when he moves the pen close to the surface, but he can't actually move his cursor or anything. In the console, where it lists all the of tablets it is trying to detect, under the CTH-480 it says [IOException:Error] SetFeature failed. When he turns his wacom drivers back on, his cursor works again, but when he disables it and tries to use the tablet driver, he can't do anything. He didn't change anything about his PC or the drivers when it stopped working. He has also tried to uninstall and reinstall the driver itself as well as completely removing it from his system and replacing it to no avail. Any thoughts?

InfinityGhost, 2021-02-03:
Are the Wacom tablet drivers installed when trying to run OpenTabletDriver?

iceflarexd, 2021-02-04:
> Are the Wacom tablet drivers installed when trying to run OpenTabletDriver?

Yes, but I also have them installed and I can use OTD with no issue, as was he before randomly it stopped working. I'm just sure to disable the wacom service before launching OTD

X9VoiD, 2021-02-04:
SetFeature error is already resolved on [v0.5.1 pre-release build](https://discord.com/channels/615607687467761684/615939905566015493/802310720314015744).

InfinityGhost, 2021-02-18:
> SetFeature error is already resolved on [v0.5.1 pre-release build](https://discord.com/channels/615607687467761684/615939905566015493/802310720314015744).

v0.5.1 is released, closing.

closed 2021-02-18 by InfinityGhost