OpenTabletDriver/OpenTabletDriver #716: Support Request: Cursor flicks to random position

TraceLosu, 2021-01-25:
# Support Request
My tablet (CTL-472) sometimes "flicks" to a random spot when my pen is moving. This only happens when hovering. I changed to HawkuDriver just to see if the problem still exists and it still had the same problem. Turning on smoothing filter or changing to the default Wacom tablet made the pen feel laggy and unresponsive so I don't want to do that. Any other fix/workaround I could do to fix this? If there isn't, how should i use the smoothing filter?

Here is a video showing the problem: https://streamable.com/scxmeo

## What I've already tried
Smoothing filter with 25ms latency and 250Hz frequency. Changing to HawkuDriver. Both didnt help. Changing to official Wacom drivers helped but the smoothing is really laggy. I've uninstalled all other drivers. 

## Diagnostic Information
https://pastebin.com/ALdYEhua

X9VoiD, 2021-01-25:
Unfortunately, there's nothing you may do for this. This is a hardware issue. Or you may use OTD's HawkuFilter (TD Noise Reduction).

TraceLosu, 2021-01-25:
How would i use the TD Noise Reduction plugin? What parameters should i start with for my tablet?

X9VoiD, 2021-01-31:
Are you also getting the same problem when using that tablet on a different computer? different usb port?

Because CTL-472 shouldn't have those issues. For the configuration of the noise reduction filter you may always just experiment to figure out what's the effect of the tunables and adjust it to your needs. The higher values you put the higher the latency.

You may close this issue now since it isn't really an issue that OTD have to fix (since this is not a normal CTL-472 behavior)

InfinityGhost, 2021-02-01:
Being that this appears to be a hardware issue, I'm closing this issue.

closed 2021-02-01 by InfinityGhost