OpenTabletDriver/OpenTabletDriver #720: OTD won't correctly refresh plugins when uninstalling and reinstalling plugins with the same filenames

Kuuuube, 2021-01-26:
## Description

OTD won't correctly refresh plugins when uninstalling and reinstalling plugins with the same filenames

To reproduce: 
1. Manually install any plugin under the same foldername and filename as a plugin in the plugin repo
(it can be an entirely different plugin with entirely different metadata as long as the foldername and filename are the same)
2. Uninstall the plugin with the plugin manager
3. Reinstall the plugin with the plugin manager

OTD now still shows the old plugin that isn't installed anymore and does not show the newly installed plugin

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Win10
| Software Version | 0.5.1 and 9ed806996b66fff557d9491f3a155a960c2768b7

X9VoiD, 2021-01-27:
For more information:

### Bug gets triggered

| | Same | Different |
| :-- | :--: | :--: |
| Folder name | x |  |
| File name | x | |

### Bug don't get triggered

| | Same | Different |
| :-- | :--: | :--: |
| Folder name | x |  |
| File name | | x |

| | Same | Different |
| :-- | :--: | :--: |
| Folder name | | x |
| File name | x | |

### Misc Info

The installed plugin is correct, but the loaded type is different.

InfinityGhost, 2021-07-18:
Does this still occur?

Kuuuube, 2021-07-18:
> 
> 
> Does this still occur?

It does.

PR #4007 (2025-09-01): Fix Plugin manager Metadata Desynchronization
This fixes an issue where even after updating the plugin, it wouldn't save the metadata to a file & the UX wouldn't load the updated metadata.

Closes #720