OpenTabletDriver/OpenTabletDriver #721: Add support for xp-pen artist 24

tabletedpolum, 2021-01-27:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    |
| Product Name    |
| Width           |
| Height          |
| Pressure Levels |
| LPI             |

## Diagnostic Information

Mrcubix, 2021-01-27:
Please fill the template, it's there for a reason

tabletedpolum, 2021-01-27:
xp pen
artiste 24 pro
23.8 inch
8192
idk what lpi is

heres their site https://www.storexppen.com/buy/artist-24-pro.html?gclid=Cj0KCQiAmL-ABhDFARIsAKywVadmPshsO9gdN_xWd0MxQyMVMGFHtqpy20J-L-FADHVVR2ctTPiR6zwaAvzJEALw_wcB

Mrcubix, 2021-01-27:
we still need the diagnostic obtained inside Of OpenTabletDriver.UX.Wpf (on windows), or .Gtk on unix based OSes

In the Ux, click on help > export diagnostic and upload it here, zip it if necessary for upload

X9VoiD, 2021-01-27:
@tabletedpolum OTD -> Help -> Export Diagnostics, upload the generated file here

jamesbt365, 2022-09-12:
Closing as this has been completely abandoned and the :bread: has gone completely stale :(

closed 2022-09-12 by jamesbt365

evertedsphere, 2023-07-08:
@Mrcubix @jamesbt365 I have the same model. 

Here is a diagnostics dump.
[artist24pro.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11990793/artist24pro.txt)