OpenTabletDriver/OpenTabletDriver #731: pen buttons do not work with H1611

NoahDTheHorse, 2021-01-29:
## Description

my pen buttons do not work when using my h1611 tablet but do work when using hs611 tablet.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 64
| Software Version | 0.5.1
| Tablet           | Huion H1611

NoahDTheHorse, 2021-01-29:
I compared their configuration files and the only real difference were the lines that said
"End": 64,
 "EndInclusive": true

so I changed them to what was in the hs6111 configuration. and it fixed it. Now it those lines are 
"End": null,
"EndInclusive": false

InfinityGhost, 2021-07-18:
The configuration for this tablet has since been modified, chances are this has been resolved in the process. If this issue still occurs, it can be reopened.

closed 2021-07-18 by InfinityGhost