OpenTabletDriver/OpenTabletDriver #740: Ability to add key combinations to the bindings

supertroll, 2021-01-31:
As it stands one can only add a single key to a binding, which makes one unable to do fairly common things like binding ctrl+z to a pen button, one _can_ use the script runner plugin to do the job, but this is pretty clunky, compared to simply binding key combinations to the tablet bindings.
This seems to me like a basic feature, which makes me wonder why no one else has requested it before.

InfinityGhost, 2021-01-31:
> This seems to me like a basic feature, which makes me wonder why no one else has requested it before.

It's been mentioned before, but it was something that we kinda forgot about and it was at the bottom of the TODO list anyway. Plan on seeing it in v0.6

Mrcubix, 2021-01-31:
As ghost said, it had been mentioned, but users also know they have access to the script runner plugin and know they can use ahk to do these dual keys combinaison

InfinityGhost, 2021-04-19:
Just realized, duplicate of #413

closed 2021-04-19 by InfinityGhost