OpenTabletDriver/OpenTabletDriver #745: Build 0.5.1 Doesn't work in Debian 10

NaokiStark, 2021-02-01:
## Description
I tried to install in Debian 10, but systemd can't run the bin (code=exited, status=203/EXEC), I tried to run binary files and doesn't work too 

[image]

I installed 0.5.0 build and works without a problem
[image]

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System |  Debian 10.7 (Buster) |
| Software Version | 0.5.1 |
| Tablet           | Wacom CTL-490 |

InfinityGhost, 2021-02-01:
Can you get the log for v0.5.1?

NaokiStark, 2021-02-01:
```ene 31 21:43:26 fabi-pcDeb systemd[2908]: opentabletdriver.service: Failed to execute command: No such file or directory
ene 31 21:43:26 fabi-pcDeb systemd[2908]: opentabletdriver.service: Failed at step EXEC spawning /usr/share/OpenTabletDriver/OpenTabletDrive
-- Subject: Process /usr/share/OpenTabletDriver/OpenTabletDriver.Daemon could not be executed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The process /usr/share/OpenTabletDriver/OpenTabletDriver.Daemon could not be executed and failed.
-- 
-- The error number returned by this process is ERRNO.
ene 31 21:43:26 fabi-pcDeb systemd[877]: opentabletdriver.service: Main process exited, code=exited, status=203/EXEC
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit UNIT has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 203.
ene 31 21:43:26 fabi-pcDeb systemd[877]: opentabletdriver.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit UNIT has entered the 'failed' state with result 'exit-code'.
```

InfinityGhost, 2021-02-01:
Hmm, that doesn't seem to give any extra useful info. Is there a file at `~/.config/OpenTabletDriver/daemon.log`? Send it here if it exists.

NaokiStark, 2021-02-01:
File not exist

InfinityGhost, 2021-02-04:
Resolved in #752

closed 2021-02-04 by InfinityGhost