OpenTabletDriver/OpenTabletDriver #749: Huion 610 Pro not detected

WhiteZhark, 2021-02-02:
I know it's on the fully functional list but it's not being detected for me. Here's the sequence of events:

Already had Huion drivers
Uninstalled Huion drivers and installed hawku driver
Uninstalled hawku drivers and installed OpenTabletDriver
Huion 610 pro not detected
Deleted "Pen Tablet" from windows devices, deleted it also from device manager (first time there was no box to delete drivers with it)
Replugged, still not detected
Tried repeatedly uninstalling the (now generic usb device) from device manager and replugging, still not detected

Something is still handling input from the tablet as I can use it as the mouse on my main monitor, but I don't know what that is

Mrcubix, 2021-02-02:
uninstall winUSB from your programs, if a windows covering half your screen appear, reinstall huion drivers and then proceed to first uninstall WinUSB and then the huion drivers, once that's done do the following steps:

Razer Synapse is known to cause issues with OTD

Device Manager -> View -> Devices by Container
Find for any of these devices:
  -  Gaomon HID/Tablet
  -  Huion HID/Tablet
  -  Graphics Tablet Device
  -  HID Tablet
Expand drop down then right click on matching device
Click Update -> "Browse my computer" -> "Let me pick" -> HID-compliant device or USB Input Device

Mrcubix, 2021-02-02:
also use this version of OTD
https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/526759335

WhiteZhark, 2021-02-02:
Reinstalled Huion Drivers, then uninstalled winUSB and Huion drivers. Current build still does not detect tablet. Haven't tested yet with the dev build due to temporary internet issues preventing download. None of my devices match the names you listed. This is how it appears in the device manager (translated):

USB Composite Device

>USB Input Device
>>HID Standard Pen
>>HID Standard Mouse

>USB Input Device
>>HID Standard Pen
>>HID Standard Vendor Defined Device
>>HID Standard Mouse

>USB Input Device
>>HID Keyboard Device
>>HID Standard Consumer Control Device
>>HID Standard System Controller

Apologies if any of that doesn't quite match standard English IT terms.

I tried installing my current non-dev build of OTD on my Windows 7laptop and it didn't detect my tablet either. I've never used a tablet before with my laptop and it had no preexisting drivers, just whatever Windows 7 installed to deal with it when I plugged it in. Main computer is Windows 10 for reference. Will report back again as soon as I'm able to test the dev build.

WhiteZhark, 2021-02-02:
Double checked with my laptop, and my tablet shows in the device manager the exact same way as my desktop: USB Composite Device with those sub-items.

Mrcubix, 2021-02-02:
In OTD : help > export diagnostic and upload it here (zip if need be)

WhiteZhark, 2021-02-02:
No luck with the H610 Pro V2 json. I don't think mine is a V2 though. I tried saving that file as the not V2 version as well just to test but it still wasn't detected. That diagnostic only has the file you offered in place of the original v2. Could this be an issue with USB 2.0 vs 3.0?
[UndetectedH610ProDiagnostic.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5908616/UndetectedH610ProDiagnostic.zip)

Mrcubix, 2021-02-02:
Just to make sure, can you screenshot and translate the content of PenTablet when doing view > device by container

WhiteZhark, 2021-02-02:
It appears multiple times, I assume because I tried plugging it into different USB ports. It's the same items, just in a different order.
[image]

HID Keyboard Device
HID Standard Consumer Control Device
HID Standard System Controller
HID Standard Pen
HID Standard Pen
HID Standard Vendor Defined Device
HID Standard Mouse
HID Standard Mouse
USB Composite Device
USB Input Device
USB Input Device
USB Input Device

Is it possible that the issue is related to my IME interacting with the tablet?

Mrcubix, 2021-02-02:
Won't lie to you, You might have been scammed by Huion and bought a "610 Pro" when in reality it's just a recycled 420 
the 420 won't be supported since it require support for a special windows driver which is also complicated to support with our current USB library (Hidsharpcore).

If you plan on buying another tablet, avoid buying the following:
- Huion 610 Pro and Pro V2
- Huion h420 and 420
- Gaomon S56K
- Most Huion manufactured tablet

WhiteZhark, 2021-02-02:
Hm, well the 420 looks like this
[image]

And mine looks like this
[image]

Are you saying the innards are a recycled 420? And were you able to deduce anything from the diagnostics file? I watched your troubleshooting video and when I used string index 201 in the string reader I got back "HVAN", which seems generic or something since it appears with many different string indexes.

WhiteZhark, 2021-02-02:
Found this comment on your video (and your response) and it matches my situation exactly:

"Hi, I was searching for the highest value for MaxInputReportLength which was 8 for me(there were also multiple of them with the same value), but when I changed the value of MaxInputReportLength to 8 on my Huion H610 Pro configuration, it didn't work. I tried with other values under manufacturer HUION and it seems that 3 works for me. Also, when editing the DeviceStrings, the string that came up when I input String Index of 201 was HVAN. But apparently there were already device strings in the configuration which are
"200": "HVAN",

"121": "HA60"
I still have the Windows Ink cursor even after adding "201": "HVAN" if that even does anything.
I also added the "200" in InitializationStrings but nothing worked.
Help pls?"

By changing MaxInputReportLength to 3, DeviceStrings to 201: HVAN and 121:HA60 I got it to detect the tablet but still with the windows ink cursor. This is with the dev version you directed me to. Is your response about USB.NET needing to be developed first before OTD can support my tablet still valid then?

X9VoiD, 2021-02-02:
I'd like to clarify some misunderstandings here.

Huion 610 Pro is not a recycled 420, but rather, it has the same firmware quirks that was first discovered on a Huion 420 tablet. The quirk is that it doesn't have any vendor interface that OTD may use to access the device input without the need for `WinUSB` (and the OEM driver for the installation of WinUSB filter driver)

> Those tablets can be accessed though with Linux and MacOS

All of the tablets of the same quirk don't follow the standard index 201 for hardware + firmware revision and index 200 for tablet initialization (disables Windows Ink).

WhiteZhark, 2021-02-02:
>Huion 610 Pro is not a recycled 420, but rather, it has the same firmware quirks that was first discovered on a Huion 420 tablet.
Thank you for the clarification. That makes sense.

To summarize here, are you saying that OTD works with the 610 Pro on Linux and MacOS but not Windows? Or that I need to reinstall WinUSB/OEM drivers for it to work on Windows?

Mrcubix, 2021-02-02:
it will work on linux and macOS since the endpoints aren't opened by the OS there

WhiteZhark, 2021-02-02:
Ok, understood. Will wait patiently for USB.NET. Thanks again for the help.

Mrcubix, 2021-09-09:
Join the [discord server](https://discord.gg/P3GYdUS) when you have time, WinUSB support has been added in pre-release and it will be faster to get it compatible if you join

gonX, 2021-12-06:
Closing as stale issue. The tablet, assuming it is the same as the Huion H610 Pro, seems to be supported in `master`.

closed 2021-12-06 by gonX