OpenTabletDriver/OpenTabletDriver #752: Pop_OS - OpenTabletDriver.UX.Gtk: No such file or directory

TheFineBroccoli, 2021-02-03:
## Description
Program Doesn't open, returns this error:

```
/usr/bin/opentabletdriver: line 10: /usr/share/OpenTabletDriver/OpenTabletDriver.UX.Gtk: No such file or directory
```
Followed installations steps for Ubuntu. Installation was error free

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System | Pop_OS! 20.04 LTS |
| Software Version | v0.5.1 |
| Tablet           | Wacom CTL-472 |

InfinityGhost, 2021-02-03:
Recompiled that release, try it again: https://github.com/InfinityGhost/OpenTabletDriver/releases/download/v0.5.1/OpenTabletDriver.deb

TheFineBroccoli, 2021-02-03:
Still same. I just tried v0.5.0 and it works

InfinityGhost, 2021-02-03:
Okay, same link. It *may* be resolved.

TheFineBroccoli, 2021-02-03:
Sorry still same. Can it be something on my side?

InfinityGhost, 2021-02-03:
Run `tree /usr/share/OpenTabletDriver` and send the output

TheFineBroccoli, 2021-02-03:
v0.5.1:

```
/usr/share/OpenTabletDriver
├── AtkSharp.dll
├── CairoSharp.dll
├── Configurations
│   ├── Gaomon
│   │   ├── 1060 Pro.json
│   │   ├── M106K.json
│   │   ├── M106K Pro.json
│   │   ├── S56K.json
│   │   └── S620.json
│   ├── Huion
│   │   ├── G10T.json
│   │   ├── GT-133.json
│   │   ├── H1060P.json
│   │   ├── H1161.json
│   │   ├── H430P.json
│   │   ├── H610 Pro.json
│   │   ├── H610 Pro V2.json
│   │   ├── H640P.json
│   │   ├── HS610.json
│   │   ├── HS611.json
│   │   ├── HS64.json
│   │   ├── Inspiroy Q11K.json
│   │   ├── Inspiroy Q11K V2.json
│   │   ├── New 1060 Plus.json
│   │   ├── osu!tablet.json
│   │   └── Q620M.json
│   ├── VEIKK
│   │   ├── A15.json
│   │   ├── A15 Pro.json
│   │   ├── A30.json
│   │   ├── A50.json
│   │   └── S640.json
│   ├── Wacom
│   │   ├── CTE-430.json
│   │   ├── CTE-440.json
│   │   ├── CTE-460.json
│   │   ├── CTE-650.json
│   │   ├── CTH-460.json
│   │   ├── CTH-470.json
│   │   ├── CTH-480.json
│   │   ├── CTH-490.json
│   │   ├── CTH-670.json
│   │   ├── CTH-680.json
│   │   ├── CTH-690.json
│   │   ├── CTL-4100 Bluetooth.json
│   │   ├── CTL-4100.json
│   │   ├── CTL-4100WL.json
│   │   ├── CTL-460.json
│   │   ├── CTL-470.json
│   │   ├── CTL-471.json
│   │   ├── CTL-472.json
│   │   ├── CTL-480.json
│   │   ├── CTL-490.json
│   │   ├── CTL-6100.json
│   │   ├── CTL-671.json
│   │   ├── CTL-672.json
│   │   ├── CTL-680.json
│   │   ├── CTL-690.json
│   │   ├── FT-0405-U.json
│   │   ├── MTE-450.json
│   │   ├── PTH-451.json
│   │   ├── PTH-651.json
│   │   ├── PTH-660 Bluetooth.json
│   │   ├── PTH-660.json
│   │   ├── PTH-850.json
│   │   ├── PTH-860.json
│   │   ├── PTK-1240.json
│   │   ├── PTK-440.json
│   │   └── PTK-540WL.json
│   └── XP-Pen
│       ├── Artist Pro 12.json
│       ├── Deco 01.json
│       ├── Deco 01 v2.json
│       ├── Deco 02.json
│       ├── Deco 03.json
│       ├── Deco mini7.json
│       ├── G430.json
│       ├── G430S.json
│       ├── G540.json
│       ├── G540 Pro.json
│       ├── G640.json
│       ├── G640s.json
│       ├── G640 v2.json
│       ├── G960 v2.json
│       ├── Star 03.json
│       └── Star 06.json
├── cs
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── de
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── es
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── Eto.dll
├── Eto.Gtk.dll
├── fr
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── GdkSharp.dll
├── GioSharp.dll
├── GLibSharp.dll
├── GtkSharp.dll
├── HidSharpCore.dll
├── ICSharpCode.SharpZipLib.dll
├── it
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── ja
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── ko
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── MessagePack.Annotations.dll
├── MessagePack.dll
├── Microsoft.Bcl.AsyncInterfaces.dll
├── Microsoft.VisualStudio.Threading.dll
├── Microsoft.VisualStudio.Validation.dll
├── Nerdbank.Streams.dll
├── Newtonsoft.Json.dll
├── Octokit.dll
├── OpenTabletDriver.Console
├── OpenTabletDriver.Console.deps.json
├── OpenTabletDriver.Console.dll
├── OpenTabletDriver.Console.runtimeconfig.json
├── OpenTabletDriver.Daemon
├── OpenTabletDriver.Daemon.deps.json
├── OpenTabletDriver.Daemon.dll
├── OpenTabletDriver.Daemon.runtimeconfig.json
├── OpenTabletDriver.Desktop.dll
├── OpenTabletDriver.dll
├── OpenTabletDriver.Native.dll
├── OpenTabletDriver.Plugin.dll
├── OpenTabletDriver.UX.dll
├── OpenTabletDriver.UX.Gtk
├── OpenTabletDriver.UX.Gtk.deps.json
├── OpenTabletDriver.UX.Gtk.dll
├── OpenTabletDriver.UX.Gtk.runtimeconfig.json
├── PangoSharp.dll
├── pl
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── pt-BR
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── ru
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── StreamJsonRpc.dll
├── System.CommandLine.dll
├── System.IO.Pipelines.dll
├── tr
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
├── WaylandNET.dll
├── zh-Hans
│   ├── Microsoft.VisualStudio.Threading.resources.dll
│   ├── Microsoft.VisualStudio.Validation.resources.dll
│   └── StreamJsonRpc.resources.dll
└── zh-Hant
    ├── Microsoft.VisualStudio.Threading.resources.dll
    ├── Microsoft.VisualStudio.Validation.resources.dll
    └── StreamJsonRpc.resources.dll

19 directories, 157 files
```

[Here](https://pastebin.com/8HUhesJh) is v0.5.0 which works

And a  diff [here](https://pastebin.com/h5hhdpYm)

InfinityGhost, 2021-02-03:
The files obviously exist, check if `OpenTabletDriver.UX.Gtk` is executable

TheFineBroccoli, 2021-02-03:
It is executable:

```
-rwxr-xr-x 1 user users 145K Jan  1  1980 /usr/share/OpenTabletDriver/OpenTabletDriver.UX.Gtk
```

Just the date seems bit odd

InfinityGhost, 2021-02-03:
> It is executable:
> 
> ```
> -rwxr-xr-x 1 user users 145K Jan  1  1980 /usr/share/OpenTabletDriver/OpenTabletDriver.UX.Gtk
> ```
> 
> Just the date seems bit odd

Appears to be a packaging issue on my end. Will update later once I get systemd containers working lol

InfinityGhost, 2021-02-04:
Issue should be resolved now

InfinityGhost, 2021-02-04:
Comment if the issue is still present, open a new issue if its fine but something new unrelated to the package arises.

closed 2021-02-04 by InfinityGhost