OpenTabletDriver/OpenTabletDriver #757: Support Request: Tablet listed as supported not found (Wacom CTL-672 on macOS Catalina 10.15.6)

gionimbus, 2021-02-04:
# Support Request

Hello everyone, 
I run macOS Catalina 10.15.6 and have downloaded OpenTabletDriver v0.5.1. When I plug in my Wacom tablet (CTL-672), the driver does not detect it. Same message shown even in the console. I have not installed any other driver on my computer. My tablet works fine (tried it on another computer). 

Thank you

X9VoiD, 2021-02-04:
OTD -> Help -> Export Diagnostics, upload the generated file here.

gionimbus, 2021-02-04:
{
  "App Version": "OpenTabletDriver v0.5.1",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "10.15.6",
    "VersionString": "Unix 10.15.6"
  },
  "Environment Variables": {
    "HOME": "/Users/giovannilupi",
    "TMPDIR": "/var/folders/64/p8bvn7sj2jn7q3brm_g9yhnc0000gn/T/",
    "SSH_AUTH_SOCK": "/private/tmp/com.apple.launchd.HMjKq1YY1Y/Listeners",
    "PATH": "/usr/bin:/bin:/usr/sbin:/sbin",
    "SHELL": "/bin/zsh",
    "XPC_SERVICE_NAME": "com.infinityghost.opentabletdriver.8248",
    "LOGNAME": "giovannilupi",
    "XPC_FLAGS": "0x0",
    "__CF_USER_TEXT_ENCODING": "0x1F5:0x0:0x4",
    "USER": "giovannilupi"
  },
  "HID Devices": [
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SPI1@15,4/AppleIntelLpssGspi@1/AppleIntelLpssSpiController@1/AppleIntelLpssSpiDevice@0/AppleHSSPIController/Apple Internal Keyboard / Trackpad/Actuator/AppleHSSPIHIDDriver",
      "VendorID": 1452,
      "ProductID": 627,
      "ReleaseNumberBcd": 1570,
      "CanOpen": true,
      "ReleaseNumber": "6.22",
      "ProductVersion": 1570,
      "Manufacturer": "",
      "ProductName": "Apple Internal Keyboard / Trackpad",
      "SerialNumber": "",
      "MaxInputReportLength": 16,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SPI1@15,4/AppleIntelLpssGspi@1/AppleIntelLpssSpiController@1/AppleIntelLpssSpiDevice@0/AppleHSSPIController/Apple Internal Keyboard / Trackpad/Device Management/AppleHSSPIHIDDriver",
      "VendorID": 1452,
      "ProductID": 627,
      "ReleaseNumberBcd": 1570,
      "CanOpen": true,
      "ReleaseNumber": "6.22",
      "ProductVersion": 1570,
      "Manufacturer": "",
      "ProductName": "Apple Internal Keyboard / Trackpad",
      "SerialNumber": "",
      "MaxInputReportLength": 5,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/CTL-672@14200000/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice",
      "VendorID": 1386,
      "ProductID": 891,
      "ReleaseNumberBcd": 256,
      "CanOpen": true,
      "ReleaseNumber": "1.0",
      "ProductVersion": 256,
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "CTL-672",
      "SerialNumber": "0LE00M1094328",
      "MaxInputReportLength": 64,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SPI1@15,4/AppleIntelLpssGspi@1/AppleIntelLpssSpiController@1/AppleIntelLpssSpiDevice@0/AppleHSSPIController/Apple Internal Keyboard / Trackpad/Accelerometer/AppleHSSPIHIDDriver",
      "VendorID": 1452,
      "ProductID": 627,
[... 526 lines cut ...]
      "Message": "Pen Bindings: [0, ], [1, ]",
      "Level": 1
    },
    {
      "Time": "2021-02-04T21:15:41.05992+01:00",
      "Group": "Settings",
      "Message": "Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]",
      "Level": 1
    },
    {
      "Time": "2021-02-04T21:15:41.059927+01:00",
      "Group": "Settings",
      "Message": "Driver is auto-enabled.",
      "Level": 1
    },
    {
      "Time": "2021-02-04T21:15:41.059935+01:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "Level": 0
    }
  ]
}

Sorry for the copy-paste but GitHub does not accept a .json file. Thank you very much.

gionimbus, 2021-02-04:
[Untitled.json.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5928291/Untitled.json.zip)

gionimbus, 2021-02-05:
Any idea about what it could be?

X9VoiD, 2021-02-05:
OTD diagnostics show that you are missing a CTL-672 endpoint, are you sure you have no other driver running in your system?

gionimbus, 2021-02-05:
Yeah pretty sure, I disinstalled the wacom drivers and also I manually checked the directories where there could be some remaining files. I didn't find any.

closed 2021-02-08 by gionimbus