OpenTabletDriver/OpenTabletDriver #758: Huion Kamvas Pro 13 (GT-133) : cursor not moving on screen

lululock71, 2021-02-05:
## Description

Hi,
After dealing with some detection issues (which https://github.com/InfinityGhost/OpenTabletDriver/issues/662 helped me a lot with it), the cursor does not move when using the stylus.

But, pen position, pen buttons and pressure is detected by OTD, because there's definitively response from the tablet when looking at the tablet debugger.

Here's my diagnostics file : [diagnostics.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5934723/diagnostics.txt)

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Arch Linux (xanmod-rt 5.10.9-rt24)
| Software Version | v0.5.1.r32.gf039db6-1
| Tablet           | Huion Kamvas Pro 13 (GT-133)

Mrcubix, 2021-02-05:
did you made any changes to the config? if yes, upload your current config here

lululock71, 2021-02-05:
Nope, config is stock. The Kamvas Pro 13 profile I created did not even work. My tablet got immediately recognized as a GT-133 (which is a profile that was already there). But the tablet outputs correctly and OTD can catch the information.

But something between processing the tablet data and moving the cursor seems to be broken...

Mrcubix, 2021-02-05:
Alright, we have a config for a different variation in PR, guess i'll change it, while we are at it, i'm going to make a config for you tablet, give me 2 min

Mrcubix, 2021-02-05:
just to make sure, go to tablet > device string reader, input the following:
9580
109
201
send request and paste teh string here

lululock71, 2021-02-05:
Apparently, the values you gave me don't work.
[image]

But I've checked in the GT-133 configuration that I have, and I found that the product ID you gave me was wrong :
[image]

I noticed that the Kamvas 13 has a productID of 109 and that the Kamvas **Pro** 13 (AKA GT-133) has 110.
These 2 models are not the same ID-wise (they both have almost the same specifications) and should not be mistaken one for another.

Mrcubix, 2021-02-05:
yeah i'm quite the pepega, figuring out some stuff

Mrcubix, 2021-02-05:
[GT-133.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5935861/GT-133.zip)
try this config

Mrcubix, 2021-02-05:
Also just to make sure, with this config, go to tablet debugger, tell me reportID when dragging and when out of range, also try to get max position

lululock71, 2021-02-06:
Hi,
I replaced the original GT-133 with your file and the tablet got successfully detected but still no cursor moving. Pen buttons and pressure are still working.

Using the tablet debugger, I have the following values :

| Value name             | Value |
| ---------------- | ----- |
| Max X | 58645
| Max Y | 33048
| Max pressure | 8191
| ReportID (out of range stylus) | 0
| ReportID (hovering with no pen buttons pressed) | 64
| ReportID (first pen button pressed) | 65
| ReportID (second pen button pressed) | 66

Also, I noticed that the 4 first buttons are registered by OTD but the slider and the fifth button (used to setup the tablet screen) are active if we look at the raw data but not registered by OTD. The fifth button is not supposed to be used as a shortcut but since it returns a press, I used it as a "free" additional shortcut key with the huion-linux-drivers script.

Mrcubix, 2021-02-06:
you sure this is the max position you can get?

lululock71, 2021-02-06:
Yes, this is the max position I can obtain, which is a little bit outside the screen frame for some reason, in the bottom right corner.

Mrcubix, 2021-02-06:
anything in the console? an error or a warning (go in debug mode and get more info there (bottom left))

Mrcubix, 2021-02-06:
[GT-133.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5937405/GT-133.zip)
try this config

lululock71, 2021-02-07:
Hi,
Just went on a trip and I bring my Arch Linux laptop and my tablet with me. For some reason, I'm back with the https://github.com/InfinityGhost/OpenTabletDriver/issues/662 issue and doing what fixed it on my desktop to my laptop does not fix the detection somehow. Might be another cause, but that means that I can't test your file as long as I have this issue...

Mrcubix, 2021-02-07:
read the wiki https://github.com/InfinityGhost/OpenTabletDriver/wiki

lululock71, 2021-02-10:
As I said before, and in issue #662, executing 
`echo "blacklist hid_uclogic" | sudo tee -a /etc/modprobe.d/blacklist.conf
sudo rmmod hid_uclogic`
on my laptop doesn't fix the issue while it does on my main machine. I don't know why and I have no time to investigate further.

InfinityGhost, 2021-07-18:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-07-18 by InfinityGhost