OpenTabletDriver/OpenTabletDriver #759: Add support for Huion Inspiron H950P

Jontique, 2021-02-05:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Huion
| Product Name    | Inspiroy H950P
| Width           | 320.8 mm
| Height          | 188.8 mm
| Pressure Levels | 8192
| LPI             | 5080

## Diagnostic Information

[H950P.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5935251/H950P.zip)

Jontique, 2021-02-05:
Tablet currently not recognized at all by the OTD. Daemon prints "[Detect:Info]   No tablet found."

Mrcubix, 2021-02-05:
in your programs, uninstall WinUSB and then huion driver if WinUSB present but not huion drivers, reinstall huion driver, uninstall WinUSB first and then uninstall huion drivers

Mrcubix, 2021-02-05:
also this:

Razer Synapse is known to cause issues with OTD

Device Manager -> View -> Devices by Container
Find for any of these devices:
  -  Gaomon HID/Tablet
  -  Huion HID/Tablet
  -  Graphics Tablet Device
  -  HID Tablet
Expand drop down then right click on matching device
Click Update -> "Browse my computer" -> "Let me pick" -> HID-compliant device or USB Input Device

Jontique, 2021-02-06:
Alright these steps helped OTD to recognize the tablet. 

Currently seems like only half of the tablet is being used to track movement horizontally, resulting in a huge increase with horizontal sensitivity (drawn circles end up as ellipses). Vertical sensitivity is working as intended. I tried all different modes but none of them fixed the issue. 

[H950P_v2.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5935901/H950P_v2.zip)

> [Settings:Info] Driver is auto-enabled.
> [Detect:Info]   Searching for tablet 'Gaomon 1060 Pro'
> [Detect:Info]   Searching for tablet 'Gaomon M106K Pro'
> [Detect:Info]   Searching for tablet 'Gaomon M106K'
> [Detect:Info]   Searching for tablet 'Gaomon S56K'
> [Detect:Info]   Searching for tablet 'Gaomon S620'
> [Detect:Info]   Searching for tablet 'Huion G10T'
> [Detect:Info]   Searching for tablet 'Huion GT-133'
> [Detect:Info]   Searching for tablet 'Huion H1060P'
> [Detect:Info]   Searching for tablet 'Huion H1161'
> [Detect:Info]   Searching for tablet 'Huion H430P'
> [Detect:Info]   Searching for tablet 'Huion H610 Pro V2'
> [Detect:Info]   Searching for tablet 'Huion H610 Pro'
> [Detect:Info]   Searching for tablet 'Huion H640P'
> [Detect:Info]   Searching for tablet ''
> [Detect:Info]   Found tablet ''

Mrcubix, 2021-02-06:
redo a diag so i can check more stuff

Mrcubix, 2021-02-06:
alright i'll be making the config, while you're here, go to tablet > device string reader and type the following:

9580
109
201
send request and paste the obtained string here

Jontique, 2021-02-06:
Device string reader output: HUION_T172_190325

Mrcubix, 2021-02-06:
[H950P.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5936141/H950P.zip)

put this config in configuration\Huion open OTD, go to tablet > tablet debugger , move your pen around ,values will appear, on the right you'll see reportID, tell me it's value when dragging and then when out of range
next, try to get max value for position.

Jontique, 2021-02-06:
Alright now it's using the whole drawing area to track 👍 

**ReportID**
While dragging: 64
Out of range: 0

Mrcubix, 2021-02-06:
also try to get max value for position, this is important verification

Jontique, 2021-02-06:
Positions max: 44200, 27600

Mrcubix, 2021-02-06:
[H950P.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5937505/H950P.zip)
try this config with set width and height

Jontique, 2021-02-06:
I set the area as **221 x 138** and it's working. 👍  Probably requires an update to the measurements in the json, but the setup seems to be working otherwise. Thank you!

Mrcubix, 2021-02-06:
you can choose whatever width and high you want on the driver unless you play full area, what i did in the json is that i set the width and height of the tablet before i sent you the new config, i'll be PRing this config so it will be available in the next release or artefacts (Action tab)

PR #761 (2021-02-18): Added Huion H950P Configuration
[image]

# Verification:

Fixes #759
closed 2021-02-18 by InfinityGhost