OpenTabletDriver/OpenTabletDriver #760: S620 Not being recognized!!! HELP

kz-n, 2021-02-06:
## Description

Gaomon S620 is not recognized, trying to re detect it doesn't do anything, if i reboot it stops working altogheter, it works on my windows pc, although that is not nearly fast enough to even run osu lol.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Manjaro Linux Nibia (Based on Arch)
| Software Version | Nibia
| Tablet           | Gaomon S620

Mrcubix, 2021-02-06:
check console  tab for errors and wiki for potential fixes

kz-n, 2021-02-06:
this is all the console says

```
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom PTH-860'
[Detect:Info]	Searching for tablet 'Wacom CTH-470'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom PTK-440'
[Detect:Info]	Searching for tablet 'Wacom CTH-460'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTL-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-672'
[Detect:Info]	Searching for tablet 'Wacom CTE-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom CTE-650'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom PTH-451'
[Detect:Info]	Searching for tablet 'Wacom PTH-660'
[Detect:Info]	Searching for tablet 'Wacom MTE-450'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom CTL-690'
[Detect:Info]	Searching for tablet 'Wacom PTH-651'
[Detect:Info]	Searching for tablet 'Wacom PTH-850'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom PTK-540WL'
[Detect:Info]	Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom XD-0608-U'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom FT-0405-U'
[Detect:Info]	Searching for tablet 'Wacom PTK-1240'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-6100'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Gaomon M106K Pro'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Huion H1161'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Huion H610 Pro V2'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Huion H610 Pro'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Huion HS610'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Huion G10T'
[Detect:Info]	Searching for tablet 'Huion GT-133'
[Detect:Info]	Searching for tablet 'Huion HS611'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Huion Q620M'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K V2'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[... 116 lines cut ...]
[Detect:Info]	Searching for tablet 'Huion 1060 Plus'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'XP-Pen G430S'
[Detect:Info]	Searching for tablet 'XP-Pen G960 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]	Searching for tablet 'XP-Pen G640 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]	Searching for tablet 'XP-Pen G640s'
[Detect:Info]	Searching for tablet 'XP-Pen G640'
[Detect:Info]	Searching for tablet 'XP-Pen Artist Pro 12'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 03'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06'
[Detect:Info]	Searching for tablet 'XP-Pen G430'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Mini7'
[Detect:Info]	Searching for tablet 'XP-Pen G540 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen Star 03'
[Detect:Info]	Searching for tablet 'XP-Pen G540'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	No tablet found.
```

kz-n, 2021-02-06:
also that anwser isn't really descripting enough @Mrcubix

Mrcubix, 2021-02-06:
https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception

kz-n, 2021-02-06:
@Mrcubix Done...
Nothing, it still has the same issue.

X9VoiD, 2021-02-06:
@kz-n The instructions in [here](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception) should have resolved the exact error that you're currently having. If it did not work for you then you most likely did something wrong, or it's already fixed but a new [error](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#argumentoutofrangeexception-value-0-15) popped up.

kz-n, 2021-02-06:
These are all the errors I get
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
I can even record myself completely removing OTD and reinstalilng everything, i may have done it 5 times already

kz-n, 2021-02-06:
^ Those errors are right after trying to detect Gaomon or Huion tablets

X9VoiD, 2021-02-06:
If it's not obvious yet, I already pointed that exact error out from my previous message predicting that you got another error. [ArgumentOutOfRange](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#argumentoutofrangeexception-value-0-15)

kz-n, 2021-02-06:
Yeah, and i followed the instructions, i still get that, i got that before and after both the first patch and this second one.

kz-n, 2021-02-06:
I retried and i got "rmmod: ERROR: Module hid_uclogic is not currently loaded"

kz-n, 2021-02-06:
looking at other repos with that same string in issue, this shouldnn't  be an error i should worry about, yet still it doesnt work
:L

X9VoiD, 2021-02-07:
Did you press redetect, or restarted the daemon after doing so? Because ArgumentOutOfRangeException *can only* happen if OTD conflicted with an active `hid_uclogic` module.

kz-n, 2021-02-07:
https://discord.com/channels/615607687467761684/628651971267788800/807881855697158144 Please take a look at this.

closed 2021-02-07 by kz-n