OpenTabletDriver/OpenTabletDriver #765: XP-Pen Artist Pro 12 Aux Parser for Auxiliary Buttons and Scrollwheel

Izerpizer, 2021-02-10:
## Description

The auxiliary buttons and scrollwheel are not detected by OTD, so they do not function. This is because there is no parser for the aux buttons on the xp-pen artist pro 12 yet. There is a missing feature card for the artist pro 12, but the parser issue isn't formally listed anywhere beyond a note that was made when the artist pro 12 was given its config file.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Ubuntu |
| Software Version | OTD v0.5.1|
| Tablet           | XP-Pen Artist Pro 12|

Izerpizer, 2021-02-10:
Not totally sure if [](https://github.com/InfinityGhost/OpenTabletDriver/labels/bug) is the right label.

InfinityGhost, 2021-02-18:
Auxiliary buttons should be doable with the GaomonReportParser (name change pending), as most other XP-Pen tablets do it that way.

gonX, 2021-12-01:
Artist 12 Pro should have 8 aux buttons supported on master now.
Please use Mrcubix' backports plugin, download a prerelease version, or wait for 0.6.

X9VoiD, 2021-12-01:
This is not resolved. The aux data is not sent on the same endpoint.

gonX, 2023-10-08:
Unrelated diag but has Artist Pro 12 endpoints:
```json
    {
      "DevicePath": "\\\\?\\hid#vid_28bd&pid_091f&mi_00&col03#8&adf8269&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "UGTABLET",
      "ProductName": "11.6 inch PenDisplay",
      "SerialNumber": "000000",
      "FriendlyName": "11.6 inch PenDisplay",
      "VendorID": 10429,
      "ProductID": 2335,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_28bd&pid_091f&mi_01#8&18f7bb59&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "UGTABLET",
      "ProductName": "11.6 inch PenDisplay",
      "SerialNumber": "000000",
      "FriendlyName": "11.6 inch PenDisplay",
      "VendorID": 10429,
      "ProductID": 2335,
      "InputReportLength": 10,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_28bd&pid_091f&mi_02#8&1342f14&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "UGTABLET",
      "ProductName": "11.6 inch PenDisplay",
      "SerialNumber": "000000",
      "FriendlyName": "11.6 inch PenDisplay",
      "VendorID": 10429,
      "ProductID": 2335,
      "InputReportLength": 12,
      "OutputReportLength": 12,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_28bd&pid_091f&mi_00&col02#8&adf8269&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "UGTABLET",
      "ProductName": "11.6 inch PenDisplay",
      "SerialNumber": "000000",
      "FriendlyName": "11.6 inch PenDisplay",
      "VendorID": 10429,
      "ProductID": 2335,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_28bd&pid_091f&mi_00&col01#8&adf8269&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "UGTABLET",
      "ProductName": "11.6 inch PenDisplay",
      "SerialNumber": "000000",
      "FriendlyName": "11.6 inch PenDisplay",
      "VendorID": 10429,
      "ProductID": 2335,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    }
```