OpenTabletDriver/OpenTabletDriver #766: Wacom CTL-4100 cursor shaking a lot and tip not working.

alixtrixia, 2021-02-12:
## Description

Recently moved from Hawku Drivers to this driver but the cursor shakes around a lot even if the pen is stationary and dragging (this never used to happen) also the tip does not work I cannot press mouse 1 with the pen.

 I tried the anti chatter filter which sorted out the shaking but also the refresh rate of my cursor very slow and it also didn't sort out the tip problem.

I've put the issues together because they occurred at the same time.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 |
| Software Version | v0.5.1 |
| Tablet           | Wacom CTL-4100 |

adryzz, 2021-03-12:
did you change the tip binding?
[image]

InfinityGhost, 2021-07-18:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-07-18 by InfinityGhost