OpenTabletDriver/OpenTabletDriver #776: Support Request: Area properties not working

SpoonMcForky, 2021-02-16:
# Support Request
 The Tablet Area settings don't do anything, with the area set up it still maps the entire tablet to all of my monitors.

[image] 
I've made my area and right clicked the upper gui to set which monitor and chose which monitor I wanted (I selected monitor 1 but that isn't even my main monitor)

## What I've already tried
 I've restarted my PC, the Wacom drivers are not running.

## Diagnostic Information

[OTDDiag.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5990398/OTDDiag.zip)

SpoonMcForky, 2021-02-16:
It's the same when it's closed, so does that mean that the tablet is not actually communicating with the driver?

Mrcubix, 2021-02-16:
if your cursor when moving your pen is a windows cursor, uninstall wacom or disable it's services

SpoonMcForky, 2021-02-16:
Also with the Wacom drivers running and OTD _uninstalled_ 2 cursers appear on the screen

Mrcubix, 2021-02-16:
double init issue fixed in dev builds https://github.com/InfinityGhost/OpenTabletDriver/actions

unplug your tablet and it will be fixed

InfinityGhost, 2021-02-25:
7 days since last response, assuming resolved.

closed 2021-02-25 by InfinityGhost