OpenTabletDriver/OpenTabletDriver #777: Support Request: Driver Stops Working When I Try to Use Windows Ink

Aiptex, 2021-02-16:
When I try to use my tablet with the windows ink plugin, the driver stops working and and displays a message (see picture below), and my cursor doesn't move when I move my pen.
[image]

I looked for any similar problems on the WindowsInk plugin wiki, but I didn't find anything.

[diagnostic.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5991144/diagnostic.txt)

Mrcubix, 2021-02-16:
first reset settings by deleting %localappdata%/opentabletdriver/settings.json

then proceed to install vmulti using the wiki link (DO NOT use hawku's vmulti) and all install steps there

X9VoiD, 2021-02-17:
Install WindowsInk's [requirements](https://github.com/X9VoiD/VoiDPlugins/wiki/WindowsInk#requirements) and proceed on doing the setup procedures if you haven't already after resetting settings.

Aiptex, 2021-02-17:
I have reset the settings and reinstalled the vmulti driver from the wiki page. I have never installed hawku's vmulti but I made sure that I have fully uninstalled them anyways, but it still comes up with the same error.

X9VoiD, 2021-02-18:
Did you install Windows Ink through the plugin manager? If you installed manually then make sure that `VMulti.dll` is within WindowsInk subfolder, not on a folder of its own.

InfinityGhost, 2021-02-25:
Stale issue after 7 days, closing.

closed 2021-02-25 by InfinityGhost