OpenTabletDriver/OpenTabletDriver #778: Multiple cursors flickering

MelonIs45, 2021-02-16:
## Description

When I tried adjusting my area, applied and then while I'm using the tablet my cursor flickers between the point it would be at if my area wasn't adjusted and the area when it is adjusted, its quite a strange behavior, I haven't installed any Wacom drivers on this system so I'm guessing its an issue with OTD.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Ubuntu 20.04
| Software Version | 0.5.1
| Tablet           | Wacom CTL-480

MelonIs45, 2021-02-16:
I have also now noticed that it happens when my cursor stays in one place for around any period of time, might help

AbstractQbit, 2021-02-16:
https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#i-have-a-ghost-cursor-constantly-alternating-cursor-position

MelonIs45, 2021-02-16:
Thank you! I should really read everything there is before submitting anything :P

closed 2021-02-16 by MelonIs45