OpenTabletDriver/OpenTabletDriver #787: Add support for Genius EasyPen i405x

Impairation, 2021-02-20:
The tablet gets detected with reportID 3 and 8 but any of these seems like not working. the tablet debugger doesn't give any output and the area doesn't changing the real area of the tablet. 

# Tablet Information
| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Genius
| Product Name    | EasyPen i405X
| Width           | 5,5"
| Height          | 4"
| Pressure Levels | 1024
| LPI             | 2540 

## Diagnostic Information

[Diagnostic i405x.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/6014026/Diagnostic.i405x.zip)

[image]

Mrcubix, 2021-02-20:
it might require an init, but i don't know what string it could be located at.
might also be wrong tho

Impairation, 2021-02-20:
i don't know if that would help https://github.com/DIGImend/digimend-kernel-drivers/blob/cb4dfa4382603b03eb3c4e26b7e3b8f049e9a425/hid-kye.c it's the same tablet there

Mrcubix, 2021-02-20:
[getstring-dump.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/6015612/getstring-dump.zip)

download this, if you have OTD running, that mean you have .net 5 installed, so extract this, and start getstring-dump.exe

on first prompt, type `1112`, on second prompt, type `20496`  and some text should print until about string 254, 

once it stopped, copy the content (click the program icon top left > edit > select all, come back there and copy instead) 

paste the content here or in a zipped notepad

Impairation, 2021-02-20:
Dump
VendorID:
1112
ProductID:
20496
Found Device
GetString (1): Genius
GetString (2): EasyPen i405X
GetString (3):
GetString (4):
GetString (5):
GetString (6):
GetString (7):
GetString (8):
GetString (9):
GetString (10):
GetString (11):
GetString (12):
GetString (13):
GetString (14):
GetString (15):
GetString (16):
GetString (17):
GetString (18):
GetString (19):
GetString (20):
GetString (21):
GetString (22):
GetString (23):
GetString (24):
GetString (25):
GetString (26):
GetString (27):
GetString (28):
GetString (29):
GetString (30):
GetString (31):
GetString (32):
GetString (33):
GetString (34):
GetString (35):
GetString (36):
GetString (37):
GetString (38):
GetString (39):
GetString (40):
GetString (41):
GetString (42):
GetString (43):
GetString (44):
GetString (45):
GetString (46):
GetString (47):
GetString (48):
GetString (49):
GetString (50):
GetString (51):
GetString (52):
GetString (53):
GetString (54):
GetString (55):
GetString (56):
GetString (57):
GetString (58):
GetString (59):
GetString (60):
GetString (61):
GetString (62):
GetString (63):
GetString (64):
GetString (65):
GetString (66):
GetString (67):
GetString (68):
GetString (69):
GetString (70):
GetString (71):
GetString (72):
GetString (73):
GetString (74):
GetString (75):
GetString (76):
GetString (77):
GetString (78):
GetString (79):
GetString (80):
GetString (81):
GetString (82):
GetString (83):
GetString (84):
GetString (85):
GetString (86):
GetString (87):
GetString (88):
GetString (89):
GetString (90):
GetString (91):
GetString (92):
GetString (93):
GetString (94):
GetString (95):
GetString (96):
GetString (97):
GetString (98):
GetString (99):
GetString (100):
GetString (101):
GetString (102):
GetString (103):
GetString (104):
GetString (105):
GetString (106):
GetString (107):
GetString (108):
GetString (109):
GetString (110):
GetString (111):
GetString (112):
GetString (113):
GetString (114):
GetString (115):
GetString (116):
GetString (117):
GetString (118):
GetString (119):
GetString (120):
GetString (121):
GetString (122):
GetString (123):
GetString (124):
GetString (125):
GetString (126):
GetString (127):
GetString (128):
GetString (129):
GetString (130):
GetString (131):
GetString (132):
GetString (133):
GetString (134):
GetString (135):
GetString (136):
GetString (137):
GetString (138):
GetString (139):
GetString (140):
GetString (141):
GetString (142):
GetString (143):
GetString (144):
GetString (145):
GetString (146):
GetString (147):
GetString (148):
GetString (149):
GetString (150):
GetString (151):
GetString (152):
GetString (153):
GetString (154):
GetString (155):
GetString (156):
GetString (157):
GetString (158):
GetString (159):
GetString (160):
GetString (161):
GetString (162):
GetString (163):
GetString (164):
GetString (165):
GetString (166):
GetString (167):
GetString (168):
GetString (169):
GetString (170):
GetString (171):
GetString (172):
GetString (173):
GetString (174):
GetString (175):
GetString (176):
GetString (177):
GetString (178):
GetString (179):
GetString (180):
GetString (181):
GetString (182):
GetString (183):
GetString (184):
GetString (185):
GetString (186):
GetString (187):
GetString (188):
GetString (189):
GetString (190):
GetString (191):
GetString (192):
GetString (193):
GetString (194):
GetString (195):
GetString (196):
GetString (197):
GetString (198):
GetString (199):
GetString (200):
GetString (201):
GetString (202):
GetString (203):
GetString (204):
GetString (205):
GetString (206):
GetString (207):
GetString (208):
GetString (209):
GetString (210):
GetString (211):
GetString (212):
GetString (213):
GetString (214):
GetString (215):
GetString (216):
GetString (217):
GetString (218):
GetString (219):
GetString (220):
GetString (221):
GetString (222):
GetString (223):
GetString (224):
GetString (225):
GetString (226):
GetString (227):
GetString (228):
GetString (229):
GetString (230):
GetString (231):
GetString (232):
GetString (233):
GetString (234):
GetString (235):
GetString (236):
GetString (237):
GetString (238):
GetString (239):
GetString (240):
GetString (241):
GetString (242):
GetString (243):
GetString (244):
GetString (245):
GetString (246):
GetString (247):
GetString (248):
GetString (249):
GetString (250):
GetString (251):
GetString (252):
GetString (253):
GetString (254):

That's the output i got.

jamesbt365, 2022-06-18:
#2238

gonX, 2022-11-07:
Completed with #2238

closed 2022-11-07 by gonX