OpenTabletDriver/OpenTabletDriver #789: Offset with Windows' rearranged displays

joonim123, 2021-02-20:
## Description

I have given my 2nd display a marginal offset(vertical) to minimize the cursor teleporting whenever it decides to travel between the two.

OTD doesn't seem to consider rearrangement, and snaps perfectly in the bottom, according to the debugger and the gui. This leads to not being able to use a portion of your [intended](https://i.imgur.com/yZ1Oxa6.png) area, if you've rearranged the displays in Windows.

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 Home 64-bit
| Software Version | v0.5.1
| Tablet           | Wacom CTL-480

InfinityGhost, 2021-02-21:
Is this a display resolution, position, or arrangement change while OpenTabletDriver is running? It currently doesn't support it without a full restart because of a lack of an event source.

joonim123, 2021-02-21:
Problem seems to be fixed by restarting system, will keep this in mind. Thanks

closed 2021-02-21 by joonim123