OpenTabletDriver/OpenTabletDriver #791: XP-Pen Star G430S not detected

Maritsu, 2021-02-21:
## Description

I plugged in my G430S, started the driver app and the drivers don't detect my tablet. I can see that they searched for it, except they didn't detect it.

## Images
[image]
[image]

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Ubuntu 20.04.2 LTS
| Software Version | v0.5.1
| Tablet           | XP-Pen Star G430S

X9VoiD, 2021-02-21:
OTD -> Help -> Export Diagnostics, upload it as a file.

Maritsu, 2021-02-22:
I can't upload, it returns me an empty file

closed 2021-03-12 by Maritsu