OpenTabletDriver/OpenTabletDriver #802: Huion h1060p not detected

abcd12345abcd, 2021-02-25:
I can move around the cursor using the tablet and pressure works, but I can't change any settings. Other then that I have no idea what is going on.

X9VoiD, 2021-02-25:
OTD -> Help -> Export Diagnostics

abcd12345abcd, 2021-02-25:
Done, what next?

X9VoiD, 2021-02-25:
Upload it here as a zipped file

abcd12345abcd, 2021-02-25:
[tablet.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/6045103/tablet.zip)

Mrcubix, 2021-02-25:
AFAIK, it should work, just to make sure, go to Tablet > Device String Reader
the type the following
9580
109
201
send request and paste the obtained string here

abcd12345abcd, 2021-02-25:
It says "HUION_T205_200714"

Mrcubix, 2021-02-25:
[H1060P.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/6045204/H1060P.zip)
extract this in Configuration/Huion

abcd12345abcd, 2021-02-25:
Thank you! I'll try this in a bit and get back.

abcd12345abcd, 2021-02-25:
...It didn't work. I really have no idea what I've done wrong.

Mrcubix, 2021-02-25:
AFAIK, if you use 0.5.1 (latest release) and this config it should work else there is a bigger issue that's not related to OTD

abcd12345abcd, 2021-02-25:
I am using both the latest release and that config, so ig something else is wrong. Thank you anyways!

Mrcubix, 2021-02-25:
just to make sure, re-upload a diagnostic with this config in the folder

abcd12345abcd, 2021-02-25:
[tablet2.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/6045377/tablet2.zip)

Mrcubix, 2021-02-25:
[H1060P.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/6045409/H1060P.zip)
One last attempt at making it work, as i know the previous config will work for other users

abcd12345abcd, 2021-02-25:
didn't work :/

Mrcubix, 2021-02-25:
go to tablet > tablet debugger, if you move your pen there, does anything happen

abcd12345abcd, 2021-02-25:
nope

Mrcubix, 2021-02-25:
just to make sure, you uninstall WinUSB in programs and features, and uninstalled vmulti if you ever installed it

abcd12345abcd, 2021-02-25:
Yeah, I did. Well, I assume I did it correctly.

Mrcubix, 2021-02-25:
well you'll have to wait for void's potential instructions cause i have no idea why your specific H1060P doesn't work rn

abcd12345abcd, 2021-02-25:
Alright. Thanks for all the help!

abcd12345abcd, 2021-02-26:
It worked! I have no idea why it wasn't working earlier, but now it's working. Thank you guys so much!

InfinityGhost, 2021-02-27:
Linking this issue to #803

closed 2021-02-27 by InfinityGhost