OpenTabletDriver/OpenTabletDriver #807: Gaomon S620 Not detected

Faalcc, 2021-02-26:
## Description

My tablet can't get detected, i tried everything, and it wont work.
I tried reinstalling, restarting and it still didn't work. Only the gaomon driver worked
[image]
[image]

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 20H2
| Software Version | OpenTabletDriver v0.5.1
| Tablet           | Gaomon S620

Faalcc, 2021-02-26:
[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Huion G10T'
[Detect:Info]	Searching for tablet 'Huion GT-133'
[Detect:Info]	Searching for tablet 'Huion H1060P'
[Detect:Info]	Searching for tablet 'Huion H1161'
[Detect:Info]	Searching for tablet 'Huion H430P'
[Detect:Info]	Searching for tablet 'Huion H610 Pro V2'
[Detect:Info]	Searching for tablet 'Huion H610 Pro'
[Detect:Info]	Searching for tablet 'Huion H640P'
[Detect:Info]	Searching for tablet 'Huion HS610'
[Detect:Info]	Searching for tablet 'Huion HS611'
[Detect:Info]	Searching for tablet 'Huion HS64'
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K V2'
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[Detect:Info]	Searching for tablet 'Huion Q620M'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom CTE-460'
[Detect:Info]	Searching for tablet 'Wacom CTE-650'
[Detect:Info]	Searching for tablet 'Wacom CTH-460'
[Detect:Info]	Searching for tablet 'Wacom CTH-470'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[Detect:Info]	Searching for tablet 'Wacom CTL-6100'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Wacom CTL-672'
[Detect:Info]	Searching for tablet 'Wacom CTL-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-690'
[Detect:Info]	Searching for tablet 'Wacom FT-0405-U'
[Detect:Info]	Searching for tablet 'Wacom MTE-450'
[Detect:Info]	Searching for tablet 'Wacom PTH-451'
[Detect:Info]	Searching for tablet 'Wacom PTH-651'
[Detect:Info]	Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom PTH-660'
[Detect:Info]	Searching for tablet 'Wacom PTH-850'
[Detect:Info]	Searching for tablet 'Wacom PTH-860'
[Detect:Info]	Searching for tablet 'Wacom PTK-1240'
[Detect:Info]	Searching for tablet 'Wacom PTK-440'
[Detect:Info]	Searching for tablet 'Wacom PTK-540WL'
[Detect:Info]	Searching for tablet 'XP-Pen Artist Pro 12'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 03'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Mini7'
[Detect:Info]	Searching for tablet 'XP-Pen G430'
[Detect:Info]	Searching for tablet 'XP-Pen G430S'
[Detect:Info]	Searching for tablet 'XP-Pen G540 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen G540'
[Detect:Info]	Searching for tablet 'XP-Pen G640 v2'
[Detect:Info]	Searching for tablet 'XP-Pen G640'
[Detect:Info]	Searching for tablet 'XP-Pen G640s'
[Detect:Info]	Searching for tablet 'XP-Pen G960 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Star 03'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06'
[Detect:Info]	No tablet found.
[Settings:Info]	Output mode: Absolute Mode
[Settings:Info]	Display area: [1920x1080@<960, 540>:0°],
[Settings:Info]	Tablet area: [0x0@<0, 0>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Ignoring reports outside area: Disabled
[Settings:Info]	Tip Binding: [Mouse Button Binding: Left]@1%
[Settings:Info]	Pen Bindings: [0, ], [1, ]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info]	Driver is auto-enabled.

Mrcubix, 2021-02-26:
first, the template asked for a diagnostic, second there is a troubleshooting tutorial as well as https://github.com/InfinityGhost/OpenTabletDriver/wiki/Windows-FAQ magic isn't it

Mrcubix, 2021-02-26:
something that isn't in the wiki however (but in my video), is the fact you need to uninstall WinUSB in your program and only then gaomon drivers.
If you don't respect that order, you'll need to reinstall gaomon driver and then uninstall winUSB and then gaomon again.

Faalcc, 2021-03-05:
i dont have winusb tho

Mrcubix, 2021-03-05:
a diagnostic is still necessary,  uninstall gaomon drivers, check devices again, as for the list of device to change, check the wiki linked above, also, check

Faalcc, 2021-03-05:
fixed it with the discord faq

InfinityGhost, 2021-03-06:
> fixed it with the discord faq

closed 2021-03-06 by InfinityGhost