OpenTabletDriver/OpenTabletDriver #816: Add support for Huion H610 Pro (2048)

HitCoder9768, 2021-03-02:
# Tablet Information
| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Huion
| Product Name    | H610 Pro
| Width           | 10"
| Height          | 6.25"
| Pressure Levels | 2048
| LPI             | 5080

## Diagnostic Information
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6065704/diagnostics.txt)
area of note
```
{
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006e&mi_01&col02#8&f943b8f&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 9580,
      "ProductID": 110,
      "CanOpen": true,
      "ReleaseNumberBcd": 0,
      "ReleaseNumber": "0.0",
      "ProductVersion": 0,
      "Manufacturer": "HUION",
      "ProductName": "PenTablet",
      "SerialNumber": "",
      "MaxInputReportLength": 0,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 6
    },
```
I attempted to implement support by hand for my install by creating a json file for its config, which caused the device to be detected but inputs don't process at all
[H610 Pro 2048.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6065715/H610.Pro.2048.txt)

In addition, do not confuse with with the newer H610 Pro that has 8192 levels of pressure sensitivity.

Mrcubix, 2021-03-02:
Your variation has a firmware that' incompatible with current our current Library, you'll need to wait for USB.NET to be released, but right now, it's still in development and isn't going out any time soon

Mrcubix, 2021-03-02:
like i said, it purely depend on the firmware, all Huion tablet might be affected but not enough people came to say exactly

HitCoder9768, 2021-03-02:
ahhh okay, thanks for letting me know.

gonX, 2021-12-01:
We have WinUSB support since #1313, there's a chance your tablet might be able to get supported now.