OpenTabletDriver/OpenTabletDriver #818: Cannot disable aspect ratio lock

adamsoltan, 2021-03-02:
## Description
After enabling area aspect ratio lock and loading settings that have it unlocked it becomes impossible to disable it (even though the checkbox isn't checked) and the area isn't the same as saved. The only way is to disable the lock is to do it before loading the settings.

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 64-bit 10.0.19042 Build 19042 |
| Software Version | 0.5.0 |
| Tablet           | Wacom CTL-472 |

X9VoiD, 2021-03-02:
Use OTD v0.5.1 instead and test again, and btw, this issue is not reproducible on v0.5.1 on my end.

adamsoltan, 2021-03-02:
I tried it with 0.5.1, still the same:

Start with unlocked aspect ratio:
[image]

Enable aspect ratio lock:
[image]

Load settings without it:
[image]

It is now locked (I changed only the width) and the only way to unlock it is to restart the driver and unlock before loading:
[image]

X9VoiD, 2021-03-02:
Ah, you mean aspect ratio lock is still applied when loading settings with unlocked aspect ratio.

adamsoltan, 2021-03-02:
> Ah, you mean aspect ratio lock is still applied when loading settings with unlocked aspect ratio.

Yes, in the loaded settings aspect ratio lock is disabled.

X9VoiD, 2021-03-02:
Rename issue to -> `Aspect ratio lock always applied on first load`

X9VoiD, 2021-03-02:
Root cause of the issue discovered. You may rename it to -> "Some context menu entries are not updated"

Affected entries:
- "Lock aspect ratio"
- "Area clipping"
- "Ignore input outside area"

InfinityGhost, 2021-07-18:
This has since been resolved since we changed to binding-based settings.

closed 2021-07-18 by InfinityGhost